KLayout/klayout #1356: Instance Property form: Array Instance col/row would be more intuitive then the current row/col

stefanottili, 2023-04-30:
In the current Instance Property form, the Array Instances entries are presented as rows/colums.
Please reverse this order, col/row (x, y) would be more intuitive..

It's also the order in which gds aref col/row number and coordinates are defined.
Oasis repetitions are defined as N-column (x) by M-row (y) dimensions.

klayoutmatthias, 2024-03-23:
This is how the three places for arrays now look like:

[image]

[image]

[image]

stefanottili, 2024-03-24:
great, 'now' as in 'upcoming', right ?
I find the col/row (x/y dimension) much more intuitive then today's master:
[image]

klayoutmatthias, 2024-03-25:
Yes, "now" means "soon" :)
I want to finalize and push out 0.29. It has been in the oven for too long already :(
Matthias

closed 2024-03-29 by klayoutmatthias