KLayout/klayout #1357: Add binding for QObject::findChildren

EugeneZelenko, 2023-05-02:
Hi, Matthias!

It'll be great to add bindings for `QObject::findChildren`. See [Patch.txt](https://github.com/KLayout/klayout/files/11369092/Patch.txt) for  my naive Qt 5 patch for `QList<T> findChildren(const QString &name = QString()) const` variant. According to Qt 4 and 6, bindings should be similar, but I had only Qt 5 to verify.

I'm not sure how hard would be `findChildren(const QRegularExpression &re) const`, but it could be skipped if it's non-trivial.

klayoutmatthias, 2023-05-03:
This code is auto-generated, so patching the file will not survive the next generation run.

The right place was inside `scripts/mkqtdecl5/mkqtdecl.conf` and correspondingly for Qt4 and Qt6. But the code generation is kind of magic as I run it against a specific golden Qt API.

EugeneZelenko, 2023-05-03:
Hi, Matthias!

See new [Patch.txt](https://github.com/KLayout/klayout/files/11391945/Patch.txt) of script changes. I read Qt 4 and 6 documentation and implementation there should be same.

closed 2023-05-23 by klayoutmatthias