KLayout/klayout #1359: Can't register package to Salt-Mine

Jan-David-Black, 2023-05-07:
I have problems to register a package to the saltmine (It is my first one, so probably a user-error).
I have been following this tutorial:

https://www.klayout.de/package_cookbook.html

According to that the link I should use for registration is

https://github.com/Jan-David-Black/paho-mqtt-klayout.git/tags/v0.0.1

That however does not seem to work. If I follow the grain.xml link I get `Not Found`...

For convenience: https://github.com/Jan-David-Black/paho-mqtt-klayout.git/tags/v0.0.1/grain.xml

[image]

klayoutmatthias, 2023-05-08:
Hi @Jan-David-Black,

sorry for that inconvenience.

I debugged the issue and as it looks, the "-" inside the package name causes this issue - contrary to what the error message says.
Could you please try to use an underscore ("paho_mqtt") instead in the "<name>" tag of "grain.xml"?

Even if that works I'd keep this ticket to remind me to fix this salt-mine server bug.

Best regards,

Matthias

klayoutmatthias, 2023-05-08:
I think, this will still not work as the paho-mqtt repository uses a subproject and the GitHub SVN bridge does not support recursive resolution. Hence, the subproject will be empty.

KLayout is not a GIT client. It talks to github via the WebDAV/svn API, nothing else.

Matthias

Jan-David-Black, 2023-05-15:
Locally it seems to work, without any additional python packages. At the time I thought it was a nice thing to use submodules to reference the original repo. It is probably easier to just copy over the code though. I'll introduce the changes tomorrow and try again!

Jan-David-Black, 2023-05-15:
The underscore fixed the publishing problem :)