KLayout/klayout #1367: Execution an error occurred during build.sh compilation, 'libklayout_rba. so: undefined reference to ` rb_gc_writebarrier_unprotect'. How can I resolve it? '

huangwenjunlovedy, 2023-05-12:
Execution an error occurred during build.sh compilation, 'libklayout_rba. so: undefined reference to ` rb_gc_writebarrier_unprotect'. How can I resolve it? '
[image]

huangwenjunlovedy, 2023-05-12:
My system is Centos7

klayoutmatthias, 2023-05-12:
This is not a log, this is a screenshot. Send the whole log and I may be able to help.

My recommendation is to build inside a Docker container. The centos7 Docker image that works for me is build using this Dockerfile:

```
FROM centos:7

RUN yum -y update \
 && yum clean all \
 && yum -y install \
      gcc \
      gcc-c++ \
      make \
      qt \
      qt-devel \
      ruby \
      ruby-devel \
      python3 \
      python3-devel \
      wget \
      fontconfig \
      gtk2 \
      dejavu-sans-mono-fonts \
      dejavu-sans-fonts \
      dejavu-serif-fonts \
      dejavu-fonts-common \
      openssh-server \
      java-1.8.0-openjdk.x86_64 \
      git \ 
      which \
      rpm-build \
      xorg-x11-server-Xvfb \
      sudo
      
COPY jenkins_id_rsa.pub .

RUN mkdir /var/run/sshd && \
    groupadd -g %JENKINS_GID% jenkins && \
    useradd -m -u %JENKINS_UID% -g %JENKINS_GID% jenkins -d /home/jenkins && \
    usermod -G wheel jenkins && \
    echo "%wheel ALL=(ALL) NOPASSWD: ALL" >>/etc/sudoers && \
    echo "jenkins ALL=(ALL) NOPASSWD: ALL" >>/etc/sudoers && \
    mkdir ~jenkins/.ssh && \
    chown jenkins ~jenkins/.ssh && \
    chgrp jenkins ~jenkins/.ssh && \
    chmod 700 ~jenkins/.ssh && \
    cat ./jenkins_id_rsa.pub >>~jenkins/.ssh/authorized_keys && \
    chown jenkins ~jenkins/.ssh/authorized_keys && \
    chgrp jenkins ~jenkins/.ssh/authorized_keys && \
    chmod 600 ~jenkins/.ssh/authorized_keys && \
    ssh-keygen -f /etc/ssh/ssh_host_ecdsa_key -N '' -t ecdsa -b 521 && \
    ssh-keygen -f /etc/ssh/ssh_host_rsa_key -N '' -t rsa && \
    ssh-keygen -f /etc/ssh/ssh_host_dsa_key -N '' -t dsa && \
    ssh-keygen -f /etc/ssh/ssh_host_ed25519_key -N '' -t ed25519 && \
    /usr/sbin/sshd
```

huangwenjunlovedy, 2023-05-16:
Based on your suggestion, I used Centos7 Docker and successfully compiled klayout. However, when running the klayout program in the container, there is no interface to display it. How can I see the program interface?
[image]

klayoutmatthias, 2023-05-16:
I meant using the Docker container you can build KLayout and deploy the resulting binaries to your target system. This way you do not have to deal with build problems arising from a contaminated system.

Just make sure you have installed the corresponding packages there or at least "qt", "ruby" and "python3".

Matthias

huangwenjunlovedy, 2023-05-17:
I copied the compiled klayout project from the Docker container to a Centos7 system outside the container. However, when I attempted to run the klayout program, the interface appeared as a blank page rather than the expected klayout interface. What could be the reason for this discrepancy?
[image]

huangwenjunlovedy, 2023-05-22:
https://github.com/KLayout/klayout/issues/1367#issuecomment-1550626276
@klayoutmatthias Could you please answer this question? Thank you very much!

chkkbim, 2023-05-22:
Did you check that you have these packages installed : qt, ruby, python3 ?

What is the output when you type these 4 commands separately in console ?
which qt
which ruby
which python3
klayout -d 20

klayoutmatthias, 2023-05-22:
I think your target system is lacking the font packages and/or the image plugins from Qt.

But without access to the system I cannot tell further details.

Matthias

klayoutmatthias, 2023-05-22:
Please try to install all the packages listed above also on your target system.

huangwenjunlovedy, 2023-05-25:
@klayoutmatthias @chkkbim Thank you very much for your answer. I have successfully run klayout!
But I can only copy the entire klayout project compiled in the Docker container to the Centos host, and finally run klayout and successfully display the interface. But I want to use X server directly in the container to display the klayout interface, but reporting errors as shown in the following figure. What should I do?

Centos host environment:
[image]
[image]

Running klayout in the Docker container reported an error:
[image]

klayoutmatthias, 2023-05-25:
Docker is a standard technology and there are many tutorials how to use it with UI applications. One I used myself is this one: https://janert.me/guides/running-gui-applications-in-a-docker-container/