KLayout/klayout #1387: "Sized" question

dsus4wang, 2023-05-24:
Hi ,
I just found the "sized" command from GUI(edit -> selection -> size shapes) is different from that function in Polygon class.
[image]

It's obviously the former is better. Could you tell me how to make the former sized transformation in code?Thanks a lot

dsus4wang, 2023-05-24:
The Picture:
1: the original polygon (has round corner)
2:using the GUI sized transformation
3: using the Polygon.sized() function

klayoutmatthias, 2023-05-25:
The description of the "size" function explains that: https://www.klayout.de/doc-qt5/code/class_Polygon.html#method78. 
Another way is to use the "Region" class instead of "Polygon".

closed 2023-05-25 by klayoutmatthias