KLayout/klayout #1402: gds file display

SouthChinaHuihuisauce, 2023-06-19:
Hello, author, I would like to consult you. I have compiled klayout in ubuntu, and now I want to transfer a gds file through http network request. I want to show through klayout where I need to find the related interface of klayout. Or where in the source code is the klayout for the gds diagram display?

klayoutmatthias, 2023-06-22:
I'm sorry, I don't understand. What exactly are you trying to achieve?

SouthChinaHuihuisauce, 2023-06-25:
> I'm sorry, I don't understand. What exactly are you trying to achieve?

Thank you for your reply, I want to find the File->Open function in the source code, but I did not find it, I would like to ask you, when in the menu bar of the klayout home page click File and then click Open where is the function, thank you
[image]

SouthChinaHuihuisauce, 2023-06-25:
> I'm sorry, I don't understand. What exactly are you trying to achieve?

After klayout analyzes the gds file, I want to know how klayout is to draw the graph and show it on the interface, I want to find the source code to learn.

SouthChinaHuihuisauce, 2023-06-25:
I have found this, and I want to know whether the function of this can meet my desire, to obtain the picture displayed by gds through http request, but is it the problem of the size of the picture? I have tried it on both win system and linux system, but the picture cannot be displayed, can I write the above code in other ides? I don't know if I can write code in another ide, such as pycharm, etc., is it only necessary to import the import
[image]

sebastian-goeldi, 2023-06-25:
I am not sure I 100% understand what you want to achieve, but are you trying to send screenshots of the layout through web connections? If so I suggest to checkout either [kweb](https://github.com/gdsfactory/kweb) which comes in the form of a fastapi application that can handle multiple sessions, or the base project from Matthias [canvas2canvas](https://github.com/klayoutmatthias/canvas2canvas) .

SouthChinaHuihuisauce, 2023-06-25:
> I am not sure I 100% understand what you want to achieve, but are you trying to send screenshots of the layout through web connections? If so I suggest to checkout either [kweb](https://github.com/gdsfactory/kweb) which comes in the form of a fastapi application that can handle multiple sessions, or the base project from Matthias [canvas2canvas](https://github.com/klayoutmatthias/canvas2canvas) . These are based on the standalone python module though, not on the full application with frontend. And they are sending screenshots by websocket not pure http(s)

Thank you for your reply, I want to do a small demo, the front-end user upload a gds file, I will use python to write the back-end, through the gds file parsing (here need to use the klayout api), no matter in what form (currently I think of the use of screenshots in the form, I do not know if there is a better way), I found macro development in klayout, but I got an error when I migrated the code to pycharm. I installed the pya library, but it doesn't seem to work, do you have any suggestions

sebastian-goeldi, 2023-06-25:
I think in that case, since I think you don't really need the GUI for this, I would suggest to build a webapp for uploading/view etc. And then use the pure python [module[(https://pypi.org/project/klayout/) for the reading and digesting of the uploaded gds.

You theoretically could probably handle it with the internal pya module, but you'd have to rely on Qt and klayout modules, which isn't ideal for a webapp imo.

SouthChinaHuihuisauce, 2023-06-25:
Thank you very much. I saw it https://pypi.org/project/klayout/ It seems very useful to me, and thank you very much for your suggestion. It made me realize that I was in the wrong direction. Thank you again, I will study Pypi carefully next, including how to read GDS files and how to display them on the page. Would you prefer to display them on a webpage or an application like QT

> I think in that case, since I think you don't really need the GUI for this, I would suggest to build a webapp for uploading/view etc. And then use the pure python [...]

SouthChinaHuihuisauce, 2023-06-25:
> I think in that case, since I think you don't really need the GUI for this, I would suggest to build a webapp for uploading/view etc. And then use the pure python [...]

Sorry to bother you again. I looked at the example and it seems that pypi is unable to parse and convert gds files into images. Am I unable to find a method, or does it really not have this method, or do you have any suggestions

SouthChinaHuihuisauce, 2023-06-25:
Simply put, I just want to convert gds files into png or other image formats, as shown in the figure. I want to know if an editor like pychar can do it in a standalone IDE. I have reviewed the relevant APIs for pypi mentioned earlier and have achieved some functions such as creating gds files, but I have not found how to convert gds files into png or other image formats
[image]

sebastian-goeldi, 2023-06-25:
https://klayout.de/doc-qt5/code/class_LayoutView.html#method103

This one has a write_png or to_png_data methods. If you need an example chdck out the kweb linked before. It uses this one.

This is probabky what you want. For the functionality you want I would go for a webapp.

If you want to use pya in an IDE you will have to set klayout as the interpreter with the appropriate flags, I think. But I'd highly recommend the standalone package

SouthChinaHuihuisauce, 2023-06-25:
[image]
I seem to have misunderstood your meaning. Please forgive me for not being familiar with how to call the API

SouthChinaHuihuisauce, 2023-06-25:
What does an independent software package mean? Can you give me an example? I want to use the Flask in Python to send HTTP requests or websocket requests, but I still haven't implemented using an independent IDE such as Pychar to call the API of Klayout to convert GDS files into PNG or other formats

sebastian-goeldi, 2023-06-25:
Yes, so you should literally do everything woth `klayout.db` and for the rest, I would really realky recommend check out kweb. It is a FastAPI implementation of a viewer that sends pngs by websocket to the client. So, it got basically everything that you ask for.

There are several ways to get from layoutview to a layout, easiest I can think of:

```python

import klayout.lay as lay

lv  = lay.LayoutView()
cvi = lv.load_layout(pathtofileasstring)

lv.active_setview_index = cvi

layout = lv.active_cellview().layout()
```

SouthChinaHuihuisauce, 2023-06-26:
> Yes, so you should literally do everything woth `klayout.db` and for the rest, I would really realky recommend check out kweb. It is a FastAPI implementation of a viewer that sends pngs by websocket [...]

Thank you for your suggestion. I will study kweb today

SouthChinaHuihuisauce, 2023-06-26:
@sebastian-goeldi Dear author, I have run this project and I am not sure if my opening method is incorrect. The things that should be displayed on my webpage are not displayed. I have checked and reported an error, which seems to be a path error. Can you give me a hint?
[image]

klayoutmatthias, 2023-06-26:
Hi, 

sorry for chiming in so late. I really recommend kweb that @sebastian-goeldi mentioned. It looks like it will do exactly what you need.

And it is already there :)

Matthias

SouthChinaHuihuisauce, 2023-06-26:
> Hi,
> 
> sorry for chiming in so late. I really recommend kweb that @sebastian-goeldi mentioned. It looks like it will do exactly what you need.
> 
> And it is already there :)
> 
> Matthias

Dear Matthias, it's okay. Yes, I successfully ran kweb. Fortunately, kweb contains exactly what I want. Thank you for your reply

klayoutmatthias, 2023-06-28:
Perfect! :) I'll close the issue if you don't mind.

closed 2023-06-28 by klayoutmatthias