KLayout/klayout #1403: Calibre rule-decks

Blebowski, 2023-06-29:
Hi,

I know it is not possible to read-in Calibre rule-decks in KLayout. Do you maybe know whether
there exists a stand-alone converter? Btw. is parsing Calibre rule-decks something that is
legally restricted by Mentor? Or if someone wants to have PDK rules for KLayout, it
needs to be written by hand?

We have a proprietary PDK, and we are slowly exploring open source EDA tools, and we
are trying to see how far we get.

mithro, 2023-06-29:
Hi @Blebowski!

I believe the issue with Calibre rule-decks is that nobody wants to lose access to the Calibre tooling (due to Mentor's monopoly position in certain parts of the silicon design workflow) and Mentor has shown a history of being very volatile.

There are a couple of companies out there (like Efabless, Mabrains & Antmicro) which can be contracted to create KLayout resources (which can be accelerated by access to existing resources) -- feel free to reach out to me@mith.ro if you need introductions. I'm sure that @klayoutmatthias could potentially offer some suggestions too.

I've been slowly working on how to build out tooling to make it easier to create complete PDKs including their rule-decks and documentation. See the resources at https://bit.ly/drc-testing https://bit.ly/build-pdk and https://bit.ly/python-pdk

Hope that is interesting! Always looking for collaborators in this space.

klayoutmatthias, 2023-06-29:
Hello @Blebowski,

even taking position here is a risk I will not take for the reasons that @mithro mentioned. Just forget the idea of a KLayout/Calibre bridge or converter - even a standalone one.

Apart from that, Calibre performance is stunning in many cases, it has many times more features and KLayout is no serious competitor in advanced applications. For that and for other technical reasons rooted in the basic concepts involved, I need to emphasize that KLayout is not a substitute for Calibre and it likely will never be.

One solution that people work on is providing DRC and LVS deck generators for multiple tools from a generic description plus some tool specific backend. The current motivation is to support for example Magic and KLayout decks from the same sources, but of course technically there is no limit and to my understanding, generating SVRF decks is not forbidden - although the respective backend probably can't be open source.

There is one constraint with this solution, namely being confined to standardized rules. I think that is no serious issue for the majority of rules used in legacy nodes, but design manuals may request checks that are outside the scope of such a standardized rule set. This only leaves the option to approximate them or to define new standard rules.

On the pro side I think that such a solution offers ways for automatic optimization (which is an important topic for KLayout) and harmonization of certain details such as the behavior in boundary cases. Plus this could eventually be the formalized electronic design manual that people ask for since decades now. I think this is the way to go finally.

Maybe being nice to Calibre above earns me some positive recognition by Mentor :) - I honestly think there could be a win-win situation like we have with GIMP vs. Photoshop or Blender vs. Adobe/Autodesk products. I mean, people using free tools for education, initial experiments or small scale / public domain projects while moving to production-grade systems for mission critical applications. After all, we are facing a serious shortage of chip design engineers and the availability of free entry-level tools for education is one key to the solution.

Matthias

Blebowski, 2023-06-30:
Hi,
@mithro @klayoutmatthias, thank you for the valuable (but little scary) insight into DRC politics. So the only way to go hypotetically forward with KLayout DRC, is to either write the conversion script on our side, or write the DRC rulesets in KLayour format.

@mithro - We might reach out to you, but we need to discuss this internally, plus we are really just starting the PD part of the design on our side, so it will take some time.

Speaking of performance, I understand Calibre is a standard, but roughly how much slower KLayout is ? Is it 10s of percents, or orders of magnitude? Usually, it is OK if the free tool takes few times more time. When PnR finishes in CI in the middle of the night, no-one cares if it takes 10 minutes of Calibre or one hour of KLayout. However, if KLayout takes 2 days to do the same set of checks, that Calibre handles in 5 minutes, that is most likely a showstopper.

@klayoutmatthias , thanks for such amazing tool. In our design team we do front-end + verification + synthesis, and we had FP/PNR/DRC/LVS during our first prototype done by external partner. KLayout was a way that we could for the first time actually see, what we taped-out :)

klayoutmatthias, 2023-07-02:
@Blebowski Please don't ask for performance benchmarks. Publishing benchmark data may be a violation of the license contract as well. This is really a serious issue.

To put it simply: **my goal is not to compete with Calibre**.

Instead I target "reasonable" performance. That is supporting research- or education-style projects on consumer hardware. Like 1M standard cell designs with reasonable memory footprint (< some GB) and reasonable run times (< 1h). 

My offer is to collaborate with open source / open IP projects who are willing to share designs on public PDKs and optimize based on these cases. There is a lot of potential I should be able to harvest this way. But there a also limitations and open designs will help to identify.

Matthias

Blebowski, 2023-07-03:
Hi @klayoutmatthias , I am sorry, I did not realize a performance comparison can be seen as license contract violation.
Anyway thanks a lot for the information you gave me. I will close the issue since its not really an issue, rathter a question.

closed 2023-07-03 by Blebowski