KLayout/klayout #1404: Editing API under Editor

SouthChinaHuihuisauce, 2023-06-30:
Dear author, I now have an API in viewer mode that allows me to call the lay and db libraries of klayout and display gds files. I would like to know if there is a Python API that can move polygons in GDS layouts in editor mode, preferably in Python. Can you find it in the API documentation? I would like to extend this feature in kweb or write a demo first
[image]
b

klayoutmatthias, 2023-07-02:
You don't need scripting for that. "move" is just another mode like "select" or "ruler". Once it is enabled, you can select objects with the mouse and move them in kweb like you do with the native application. Note that in order to be able to move polygons, the view needs to be opened in editable mode.

Matthias

SouthChinaHuihuisauce, 2023-07-03:
I tried to move the rectangles in kweb and canvas2canvsas, but it didn't work. Is there any other API in klayout.lay or klayout.db that can call the mode to enable movement?

SouthChinaHuihuisauce, 2023-07-03:
> You don't need scripting for that. "move" is just another mode like "select" or "ruler". Once it is enabled, you can select objects with the mouse and move them in kweb like you do with the native [...]
Dear author, I tried to choose the moving mode, but I was not successful. I want to move polygons in kweb, which uses canvas view display. It belongs to editable view display and should be movable in principle
[image]

klayoutmatthias, 2023-07-03:
Hello,

I debugged the issue myself and found a few bugs that prevent use of edit mode in the Python module: first, the layouts inherit the application's editable flag (which is not there). Instead they should inherit the view's editable mode. Second, a segfault is observed as the Undo/Redo feature is not available by default.

It's possible to work around these problems, but I think it's wiser to wait for the 0.28.10 release which is to come soon.

Matthias

sebastian-goeldi, 2023-07-03:
@SouthChinaHuihuisauce sorry didn't have time yet to test it out, but I suspected that it might be some bugs in KLayout (tested it when the standalone canvas capabilities were fresh off the press, and it crashed kweb then ;)). Also, please keep in mind I didn't intend kweb to be a full editor, I actually thought I fully disabled editabke mode. But once it's ready, I think I will build https://github.com/gdsfactory/metainfo-ports directly into kweb.

Thanks @klayoutmatthias for the fast debugging. could you link the commit(s) or PR knce you have them? I would be happy to debug and test.

klayoutmatthias, 2023-07-04:
@sebastian-goeldi I will soon merge them to master. There are two of them.

SouthChinaHuihuisauce, 2023-07-04:
Thank you for your reply. Looking forward to your next version, I will continue to study it as well

closed 2023-08-09 by klayoutmatthias