KLayout/klayout #1413: Mac OS Ventura and klayout command line

rviktor85, 2023-07-05:
Hello, I am trying to start DRC check from my external Python application via KLayout API. 

According to my research concerning this topic, it is not possible for now, and the only possibility is to run the DRC check via the klayout command line or via KLayout GUI. I tried to run via the command line, but, this command is not recognized on my computer (mac osx Ventura), and I was not able to find the binary file for the layout or to find some solution. I installed KLayout using
`brew install klayout`

Is there any solution for this or maybe for running DRC check using the API from the python code?

sebastian-goeldi, 2023-07-05:
Afaik it is possible to run a drc script from python. This class should be able to do that for you https://klayout.de/doc-qt5/code/class_Macro.html

Unless I am mistaken this is available in the python standalone package

klayoutmatthias, 2023-07-05:
No, I'm sorry. Running DRC scripts needs the Ruby interpreter and it is not available in the Python module.

The only option to run DRC scripts is through the KLayout application (`klayout -b -r yourscript.drc`) or the stripped-down version "strmrun" buddy tool. 

It is basically possible to get all DRC features inside Python scripts, because DRC is just a wrapper around certain classes (`Region`, `Edges`, `EdgePairs` etc.). But DRC will evolve further (requiring different methods and maybe classes) and the Ruby DSL wraps the scheme in form that is more compact and easy to read.

Matthias

sebastian-goeldi, 2023-07-05:
Thanks Matthias. Sorry, yeah , I somehow missed that it's in lay, and makes totally sense when thinking about it :facepalm: .

Regarding the brew installation:

I am not entirely sure, as I barely ever use MacOS, and the Mac builds are done by a community member. I think the brew installation is done by someone else. I can check on my mac test machine at some point, but currently I am quite busy. Did uou do the installation as the brew docs say? `brew install --cask klayout`. If yes, maybe their latest maintainer of the brew package can help @krehel

rviktor85, 2023-07-06:
Thank you very much for your help, @klayoutmatthias @sebastian-goeldi. I'll look a bit more into why the command line is not working on my Mac.

 I also tried installing KLayout on Ubuntu, and everything is working fine there.

Kazzz-S, 2023-08-01:
Sorry about the late response.

```
(base) MacBookPro2{kazzz-s} zzz (1)% brew install --cask klayout

==> Downloading https://formulae.brew.sh/api/cask.jws.json
################################################################################################################################################################### 100.0%
==> Downloading https://raw.githubusercontent.com/Homebrew/homebrew-cask/19a54e0e87f8a9481e481e9508b194b7ec2d3bd9/Casks/klayout.rb
################################################################################################################################################################### 100.0%
==> Downloading https://www.klayout.org/downloads/MacOS/HW-klayout-0.28.10-macOS-Monterey-1-qt5Brew-RsysPhb39.dmg <===== (1)
################################################################################################################################################################### 100.0%
==> Installing Cask klayout
==> Moving App Suite 'KLayout' to '/Applications/KLayout'
🍺  klayout was successfully installed!
```
(1) is the DMG I built.

----

```
(base) MacBookPro2{kazzz-s} KLayout (2)% pwd
/Applications/KLayout

(base) MacBookPro2{kazzz-s} KLayout (3)% ll
total 0
drwxr-xr-x   4 kazzz-s admin  128  8  1 22:30 .
drwxrwxr-x 176 root     admin 5632  8  1 22:30 ..
drwxr-xr-x  16 kazzz-s staff  512  7 11 06:15 Homebrew-HUser-ReadMeFirst
drwxr-xr-x   3 kazzz-s staff   96  7 10 21:23 klayout.app <===== (2)
```
(2) **Open this application bundle from the Finder, ignoring any security warnings.**

----

```
(base) MacBookPro2{kazzz-s} Contents (4)% pwd
/Applications/KLayout/klayout.app/Contents

(base) MacBookPro2{kazzz-s} Contents (5)% ll
total 8
drwxr-xr-x  9 kazzz-s staff  288  7 10 21:24 .
drwxr-xr-x  3 kazzz-s staff   96  7 10 21:23 ..
drwxr-xr-x 13 kazzz-s staff  416  7 10 21:23 Buddy <===== (3)
drwxr-xr-x 77 kazzz-s staff 2464  7 10 21:25 Frameworks
-rwxr-xr-x  1 kazzz-s staff 1451  7 10 21:23 Info.plist
drwxr-xr-x  8 kazzz-s staff  256  7 10 21:25 MacOS
-rwxr-xr-x  1 kazzz-s staff    9  7 10 21:22 PkgInfo
drwxr-xr-x 13 kazzz-s staff  416  7 10 21:24 PlugIns
drwxr-xr-x  4 kazzz-s staff  128  7 10 21:24 Resources
```
(3) The buddy command line tools are here.

----

```
(base) MacBookPro2{kazzz-s} Buddy (6)% pwd
/Applications/KLayout/klayout.app/Contents/Buddy <===== (4)

(base) MacBookPro2{kazzz-s} Buddy (7)% ll
total 748
drwxr-xr-x 13 kazzz-s staff   416 Jul 10 21:23 .
drwxr-xr-x  9 kazzz-s staff   288 Jul 10 21:24 ..
-rwxr-xr-x  1 kazzz-s staff 66552 Jul 10 21:24 strm2cif
-rwxr-xr-x  1 kazzz-s staff 66552 Jul 10 21:24 strm2dxf
-rwxr-xr-x  1 kazzz-s staff 66552 Jul 10 21:24 strm2gds
-rwxr-xr-x  1 kazzz-s staff 66584 Jul 10 21:24 strm2gdstxt
-rwxr-xr-x  1 kazzz-s staff 66552 Jul 10 21:24 strm2mag
-rwxr-xr-x  1 kazzz-s staff 66552 Jul 10 21:24 strm2oas
-rwxr-xr-x  1 kazzz-s staff 66552 Jul 10 21:24 strm2txt
-rwxr-xr-x  1 kazzz-s staff 66552 Jul 10 21:24 strmclip
-rwxr-xr-x  1 kazzz-s staff 66552 Jul 10 21:24 strmcmp
-rwxr-xr-x  1 kazzz-s staff 66552 Jul 10 21:24 strmrun
-rwxr-xr-x  1 kazzz-s staff 66552 Jul 10 21:24 strmxor

(base) MacBookPro2{kazzz-s} Buddy (8)% ./strmrun -h
Usage:

  strmrun  [options]  <script>

    This program runs Ruby or Python scripts with a subset of KLayout's
    API.

Arguments:

  <script>               The script to execute

        This script will be executed by the script interpreter. The script
        can be either Ruby (".rb") or Python (".py").

Options:

  Options can be specified in a short (with one dash) or a long form
  (with two dashes). If a value is required, it can be specified either
  as the following argument or added to the option with an equal sign
  (=).

  List of options:

    -d|--debug-level       Sets the verbosity level

          The verbosity level is an integer. Typical values are:
          * 0: silent
          * 10: somewhat verbose
          * 11: somewhat verbose plus timing information
          * 20: verbose
          * 21: verbose plus timing information
          ...

    -h|--help              Shows the usage and exits

    -v|--var=name=value    Defines a variable

          When using this option, a global variable with name "var" will be
          defined with the string value "value".

    --help-all             Shows all options (including advanced) and exits

    --license              Shows the license and exits

    --version              Shows the version and exits
```
(4) Set the PATH environment variable to include this directory.

lighteningq, 2023-12-26:
hey [matthias](https://github.com/klayoutmatthias) and [Kazzz-S](https://github.com/Kazzz-S), 
Is the buddy tool the only option for klayout to run CLI in mac?
I have tried `klayout -b ` via terminal but it is giving me command not found. (P.S: I have already added the PATH)
```
export PATH="/Applications/klayout.app/Contents/Buddy:$PATH"
export PATH="/Applications/klayout.app:$PATH"
```

lighteningq, 2023-12-26:
also when i try to run `strmrun mydrcscripr.lydrc`, it is giving me the following error:
```
ERROR: Can't run macro (no interpreter): /Volumes/workplace/klayoutdrc/src/Current DRC/DRC_1.2.0/DRC_run.lydrc
```

any ideas?

Kazzz-S, 2023-12-27:
Hello @lighteningq,

I'm at work right now, so I can't try this on my Mac, but my idea is to create a **`launch service script`**.
The $@ in the last line should take parameters from the command line (like -b, etc).
[image]

Below is a sample file contained in the `HomebrewUser-ReadMeFirst` folder of `LW-klayout-0.28.14-macOS-Ventura-1-qt5Brew-Rhb32Phb311.dmg`.
If you want to use it as a standalone bash script,
* In the first line, add `#! /bin/bash`
* Set execute permissions on the file

```
#---------------------------------------------------------------------------------------
# Bundle: KLayoutHomebrew.app
#
# Descriptions:
#  This script invokes "klayout.app" that shares the Homebrew development environment.
#  More precisely, it uses the Qt5, Ruby, and Python packages of Homebrew installed in
#  the standard location.
#---------------------------------------------------------------------------------------

#--------------------------------------------------------------------------
# Localizing the language
# Workaround discussed in https://github.com/KLayout/klayout/issues/1213
#--------------------------------------------------------------------------
export LANG=$(defaults read -g AppleLocale).UTF-8

#--------------------------------------------------------------------------
# Your setups
#--------------------------------------------------------------------------

#--------------------------------------------------------------------------
# Export the 'KLAYOUT_GIT_HTTP_PROXY' environment variable if you need to
# use the git-based Salt Package Manager through a proxy server.
#--------------------------------------------------------------------------
#export KLAYOUT_GIT_HTTP_PROXY="http://111.222.333.444:5678"

#--------------------------------------------------------------------------
# Export the 'PYTHONHOME' environment variable if required
#--------------------------------------------------------------------------
#export PYTHONHOME=

#--------------------------------------------------------------------------
# With "-n" option, you can invoke multiple instances
#--------------------------------------------------------------------------
myklayout=/Applications/klayout.app
myconfig=$HOME/.klayout/klayoutrc

open -n -a $myklayout --args -e -c $myconfig -style=fusion $@
```

The image and the script are out of sync. Sorry about that.

Best regards,
Kazzz-S