KLayout/klayout #1414: Feature request: Viewer option to merge all shapes

Partmedia, 2023-07-05:
Certain other layout applications have a viewer option to "merge all shapes" that I would find useful in KLayout. This is useful because sometimes cells are drawn with individual shapes that overlap when put together to form a final shape.

This viewer option is equivalent to taking all visible layers and Edit -> Layer -> Merge with itself with flat hierarchy mode, but it does not modify the underlying shapes in the GDS.

A workaround to achieve this effect is to turn off border decorations in the layer toolbox style menu.

If this is something that would be accepted conceptually, I'd be happy to attempt a PR if you have a rough idea of how this could be implemented.

klayoutmatthias, 2023-07-05:
I think turning off the border is the only option that does not come with a considerable computing overhead. Merging is a computationally expensive operation in polygon space.

What specific applications are you referring to? I have not seen a corresponding option in GDS/VLSI layout tools.

Partmedia, 2023-07-05:
This is available in Cadence Virtuoso.

**Without Merge Shapes**
[image]

**With Merge Shapes**
[image]
[image]

I wonder if it's actually doing a true merge operation or doing something else at the display level?

klayoutmatthias, 2023-07-06:
Thanks for pointing me to this. I did not know (or use) this feature.

I appears as if they are doing some kind of polygon merge, but it will also leave path center lines intact. After all I see many artefacts (as often) and drawing is slow there, so this appears a kind of compromise.

I see two options for KLayout:

* True polygon merge. That probably isn't difficult to implement although it will not have the center lines of paths and there needs to be some fallback to non-merged in case too many shapes are encountered - otherwise you will see a severe memory and runtime impact with larger layouts. KLayout promises true layout view without artefacts and it will not be possible to keep up this promise in dense layout cases. 
* Image postprocessing. I should be possible to identify pixels at the border of a polygon and highlight them as "frame" pixels instead of drawing the shape frames. This is a fast implementation and should not have a noticeable performance impact. However, it is not possible to generate a high fidelity frame pattern - for example, small gaps which are not resolved into black pixels will not receive a frame border.

Matthias