KLayout/klayout #1419: No Gui

jerry9055, 2023-07-12:
I download and install by command line in Ubuntu 20.04.3 LTS
`$ sudo dpkg -i klayout_0.28.10-1_amd64.deb `
When the installation is over
'$ klayout -v'
'KLayout 0.28.10'
But when I key in 'klayout', no GUI appear from mobaXterm
[image]

Would you do me a favour?
I am not familiar with linux

chkkbim, 2023-07-12:
[1] Is this the 1st time you are building KLayout (any version) from source code on any OS ?

[2] Could you refer to this [https://github.com/KLayout/klayout/issues/1367#issuecomment-1550124136](url) ?

jerry9055, 2023-07-13:
@chkkbim 
Thanks for your advice.
I have rebuilt Klayout maybe 4 or 5 times after it does not work. 
After reading [2], I try `klayout -d 20`, and it shows the same warning before:
`Warning: This plugin does not support propagateSizeHints()`.

This is the warning that makes klayout suck, I think. But I forget when this modification happened leading to this.
Thank you for your assistance again

jerry9055, 2023-07-16:
Today I follow this 
https://forum.qt.io/topic/104437/qt-qpa-plugin-could-not-load-the-qt-platform-plugin-xcb-in-even-though-it-was-found/8
and try
`sudo apt install libxcb-icccm4 libxcb-image0 libxcb-keysyms1 libxcb-render-util0`
Successfully open the GUI again.
Thanks for your attention.

closed 2023-07-16 by jerry9055