KLayout/klayout #1422: dxf file parsing error, about spline curve

libaineu2004, 2023-07-14:
I found a problem with parsing errors on dxf's spline curve. Here uploaded 3 dxf files, you can use klayout and AutoCAD for comparison. You can see that there are problems with spline curve analysis, either with more lines or less lines.

https://github.com/libaineu2004/myepolltest/releases/download/dxf/bo.dxf
https://github.com/libaineu2004/myepolltest/releases/download/dxf/potato.dxf
https://github.com/libaineu2004/myepolltest/releases/download/dxf/R.dxf

klayoutmatthias, 2023-07-15:
I do not have AutoCAD for comparison. Could you paste some reference images and describe the expected output?

klayoutmatthias, 2023-07-15:
I used an online viewer, so now your files are on their servers. I guess you don't care ...

I see the difference here:

KLayout:
[image]

Viewer:
[image]

So I think this is what you mean.

klayoutmatthias, 2023-07-15:
I debugged the problem and I think I fixed it. Reason was an exceptionally high point density in certain segments.

Here are my outputs now:

"R":
[image]

"potato":
[image]

"bo":
[image]

I could not find an online viewer that displays "R" and "potato", so maybe you can confirm that now that is the correct rendering.

Matthias

klayoutmatthias, 2023-07-15:
Ok, found the issue. Potato looks like this now:

[image]

I guess that fixes it.

libaineu2004, 2023-07-15:
> online viewer

What is the web address of the online viewer you are visiting? What is the url?

libaineu2004, 2023-07-15:
klayout version
0.28.8

"R.dxf" is:
[image]

klayoutmatthias, 2023-07-15:
I tried that online viewer: https://products.groupdocs.app/viewer/total, but also others had difficulties with "R" and "potato".

Anyway, if the "R" with the circular frame is correct it looks like I have fixed the problem.

klayoutmatthias, 2023-07-15:
This one (https://3axis.co/dxf-viewer/dxf-viewer/) shows the "R" like KLayout now:

[image]

But the potato looks strange:

[image]

I guess nobody is perfect ...

libaineu2004, 2023-07-16:
> Anyway, if the "R" with the circular frame is correct it looks like I have fixed the problem.

autocad is correct. Also, qcad is correct.
https://qcad.org/en/

klayoutmatthias, 2023-07-16:
Thanks for pointing me to qcad ... I tried FreeCAD, but it did not like the splines.

libaineu2004, 2023-07-20:
Have all the problems been solved? Can you tell me what has been changed in the source code?
thank you

klayoutmatthias, 2023-07-20:
Code change is thias :https://github.com/KLayout/klayout/pull/1423/commits/35e42a81178aa3fe071a7f656a4c026eeb596679

The PR is linked tovthias ticket and will be merged soon into master.

libaineu2004, 2023-07-21:
> Code change is this :[35e42a8](https://github.com/KLayout/klayout/commit/35e42a81178aa3fe071a7f656a4c026eeb596679)
> 
> The PR is linked to this ticket and will be merged soon into master.

I tried using PR's new code and found that the program would blink.
https://github.com/KLayout/klayout/commit/35e42a81178aa3fe071a7f656a4c026eeb596679

I'm already in the function spline_interpolate. Added：
assert(pe ! = curve_points.end());
[image]

The assertion will take effect and an error will be reported.

I opened the following three dxf files with the same problem. My programming environment is VS2019 x64.
bo.dxf
potato.dxf
R.dxf

klayoutmatthias, 2023-07-21:
I am using exactly these files as unit test cases and they work in my case.

libaineu2004, 2023-07-21:
> 
yes, debug mode

klayoutmatthias, 2023-07-21:
Please use the wip2 branch tip. The code reference was just for your information, not for reproducing the solution. 

wip2 branch works for me on Linux and Windows.

Find the golden data for the tests here: https://github.com/KLayout/klayout/tree/wip2/testdata/dxf (issue_1422a_au.gds.gz to issue_1422c_au.gds.gz).

Matthias

libaineu2004, 2023-07-22:
> potato.dxf

Could you just tell me. For these three dxf files, which source files have the source code changed? Besides dbDXFReader.cc, are there any other files that need to be modified?

I look at the https://github.com/KLayout/klayout/commit/35e42a81178aa3fe071a7f656a4c026eeb596679, only dbDXFReader. Cc file is modified. But the potato.dxf problem was not solved.

klayoutmatthias, 2023-07-22:
I'm sorry, but https://github.com/KLayout/klayout/pull/1423/commits/35e42a81178aa3fe071a7f656a4c026eeb596679 are the only changes made. It should be enough to apply the changes of dbDXFReader.cc - the other modifications are only new tests to cover your case. The potato is reproduced without the sticks on MSVC2017, CentOS7 and Ubuntu 22 for me.

klayoutmatthias, 2023-07-22:
I've put in an assertion and I can reproduce the array out of bounds issue. I will debug it.

klayoutmatthias, 2023-07-22:
I have placed one more patch: https://github.com/KLayout/klayout/pull/1423/commits/99df15a5fff7fd6ac0f64b781d34cb2b2a4bc706 - this should fix the potato issue.

libaineu2004, 2023-07-23:
> I have placed one more patch: [99df15a](https://github.com/KLayout/klayout/commit/99df15a5fff7fd6ac0f64b781d34cb2b2a4bc706) - this should fix the potato issue.

Yes.
https://github.com/KLayout/klayout/commit/99df15a5fff7fd6ac0f64b781d34cb2b2a4bc706
The result of my test is correct.
Thank you!

klayoutmatthias, 2023-07-23:
Very good. I'll merge the PR then.

Matthias

closed 2023-08-08 by klayoutmatthias