KLayout/klayout #1424: I would like to ask a question. Where can I find the source code for parsing and displaying GDS files in Klayout? Could you please tell me? Thank you.

bboyguan, 2023-07-17:
I would like to ask a question. Where can I find the source code for parsing and displaying GDS files in Klayout? Could you please tell me? Thank you.

[image]

stefanottili, 2023-07-18:
find src | grep cc | grep gds
find src | grep cc | grep View
both should give you a start at looking.
The code reading/writing various file formats can be found in src/plugins/streamers

To have a look at unfamiliar code, I found it help fuel to use
"valgrind --tool=callgrind klayout"
which generates callgraphs that can be visualized with [kq]cachegrind.

stefanottili, 2023-09-13:
This can be closed,  o ?