KLayout/klayout #1427: Individual Polygon Colouring

das-dias, 2023-07-24:
Hello! Thank you for this amazing tool! I was wondering if there is a way of ignoring the layer colouring presets for each layer, and attribute an individual colour for each polygon of a layer/cell. For example, if I have an {index,layer : polygon_colour} map (a simple python dictionary, for example), how can I show these colours in the KLayout viewer?

Kazzz-S, 2023-07-24:
You can create a `layer property file` (an XML file) to customize the layer-wise fill color, frame color, stipple, etc.

[sample-1427.zip](https://github.com/KLayout/klayout/files/12145508/sample-1427.zip)

First, open the attached OASIS file `inv-DMOS4.oas` with the default/preset layer properties.
[image]

Next, open the attached `inv-DMOS4.lyp` by **File -> Load Layer Properties** or just drag and drop the file to the canvas.
[image]

das-dias, 2023-07-24:
Hello! Thanks for the answer! Yes I know about the editing of the colour settings of each layer. What I was asking was if these settings can be ignored, and attribute a different colour for each of the polygons in the Layout improted from the GDS / OAS file, provided you have a map of {polygon ID :  polygon_colour}. Notice that I am talking about polygon-wise colouring, not layer-wise colouring.

Kazzz-S, 2023-07-24:
I understood; not {index,**layer** : polygon_colour} map but {**polygon ID**: polygon_colour}.

Many years ago, I had a similar need. In that case, I mapped different polygons to different L:D combinations and created a layer property file (by Python) accordingly.

klayoutmatthias, 2023-07-24:
@das-dias BTW: there is a scheme called "coloring" for multiphase masks or multi patterning lithography. It is employed in high-end nodes to extend the lithography range. You're not referring to that, are you?

If you do, the same approach applies. "colors" (in the sense of different parts of the mask) have to be separate into different layers.

closed 2023-08-09 by klayoutmatthias