KLayout/klayout #1430: Add option to filter out geometric primitives from Marker Database Browser Info widget

EugeneZelenko, 2023-07-24:
It makes sent to have possibility to  filter out geometric primitives (included in `value`) from Marker Database Browser Info widget. Main purpose of geometrical primitives is to show them in layout view and they only interfere with `text`.

klayoutmatthias, 2023-07-29:
The solution is a new switch here:
[image]

When unchecked, shapes are not listed:
[image]

The setting is persisted and can be configured through the "rdb-list-shapes" configuration property.

EugeneZelenko, 2023-07-29:
Hi, Matthias!

Thank you for fast implementation! But could this setting be controlled via configuration and be persistent there?

closed 2023-07-30 by klayoutmatthias