KLayout/klayout #1432: strm2oas def ignores path segment after + VIRTUAL + RECT

stefanottili, 2023-07-28:
strm2oas ignores path segment after + VIRTUAL .. + RECT ..

See the attached [test_virtual.tgz](https://github.com/KLayout/klayout/files/12202572/test_virtual.tgz)
```
VERSION 5.8 ;
DESIGN test ;
UNITS DISTANCE MICRONS 1000 ;
DIEAREA ( -1 -1 ) ( 8 8 ) ;
NETS 1 ;
- dummy + ROUTED M1 ( 0 0 ) ( 5 0 ) VIRTUAL ( 7 1 ) RECT ( -3 0 -1 2 ) ( 7 7 ) ;
END NETS
END DESIGN
```

klayoutmatthias, 2023-07-29:
Thanks, perfect!

I'll take a look.

Matthias

klayoutmatthias, 2023-07-29:
I basically think it is a more general topic.

How is that understood: 

```
( 7 1 ) RECT ( -3 0 -1 2 ) ( 7 7 )
```

Is that a segment `( 7 1 ) -> ( 7 7 )` and a separate rectangle? 

In other words: are rectangles objects that terminate a path are these out-of-band items and the path will continue across the rectangle?

Matthias

stefanottili, 2023-07-29:
see http://coriolis.lip6.fr/doc/lefdef/lefdefref/lefdefref.pdf page 267/268 figure 4-5 for a detailed explanation.

closed 2023-08-01 by klayoutmatthias