KLayout/klayout #1433: fix def syntax error in testdata/lefdef/viasize2/test.def

stefanottili, 2023-07-29:
test.or:
read_lef testdata/lefdef/viasize2/test.lef
read_def -continue_on_errors testdata/lefdef/viasize2/test.def

openroad test.or

ERROR (DEFPARS-5500): Def parser has encountered an error in file testdata/lefdef/viasize2/test.def at line 31, on token MASK.

diff klayout/testdata/lefdef/viasize2/test.def test_fixed.def
31c31
<   + ROUTED M1 30 + MASK 2 ( 0 0 15 ) ( 100 0 0 ) VIA1_small ( 100 100 10 )
---
>   + ROUTED M1 30 ( 0 0 15 ) MASK 2 ( 100 0 0 ) VIA1_small ( 100 100 10 )

klayoutmatthias, 2023-07-29:
Is this really a KLayout error? This message ("DEFPARS-5500") is not mine.

Matthias

stefanottili, 2023-07-29:
The syntax error is from the openroad def 5.8 si2 parser.Klayout wrongfully reads it without any errors.

klayoutmatthias, 2023-07-30:
You're right, the DEF sample was kind of nonsense and this error can be captured.

In general however, the DEF parser is sloppy (means: not always based on a formal syntax) and will accept some invalid DEF files without error. Accepting stray "+" is one example. In other cases I simply skip sections. Hope is that this way newly introduced keywords will not make the reader fail automatically.

Matthias

commit 42baaef0 (2023-08-08): Fixed #1433 (def syntax error not captured)
commit a29bef49 (2023-08-08): Merge pull request #1456 from KLayout/issue-1433

Fixed #1433 (def syntax error not captured)
closed 2023-08-02 by stefanottili