KLayout/klayout #1437: import klayout met ModuleNotFoundError: No module named 'klayout.tlcore'

Alexjiangxiaohui, 2023-07-31:
import klayout
---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-39-bbd6d8b74084> in <module>
----> 1 import klayout

/nfs/PEG/ITE/xiajiang/SMO_tool/klayout/__init__.py in <module>
      1 
----> 2 from .tl import __version__
      3 

/nfs/PEG/ITE/xiajiang/SMO_tool/klayout/tl/__init__.py in <module>
----> 1 import klayout.tlcore
      2 from klayout.tlcore import *
      3 
      4 __all__ = klayout.tlcore.__all__

ModuleNotFoundError: No module named 'klayout.tlcore'

klayoutmatthias, 2023-07-31:
I'm sorry - I don't know what this relates to. 
Is that the klayout module on PyPI? If so, there is no trouble with importing klayout.

Matthias

ThomasHilger, 2024-11-22:
In the pypi version 0.29.8 you can not import the klayout package! This is exactly what OP was trying to tell.
if you do:
```[python]
import pya
```
You will get:
```
import pya  # type: ignore[import-untyped]
..\..\..\AppData\Local\anaconda3\envs\design\Lib\site-packages\pya\__init__.py:3: in <module>
    from klayout.pya import __all__
..\..\..\AppData\Local\anaconda3\envs\design\Lib\site-packages\klayout\__init__.py:2: in <module>
    from .tl import __version__
..\..\..\AppData\Local\anaconda3\envs\design\Lib\site-packages\klayout\tl\__init__.py:2: in <module>
    from ..tlcore import __all__
E   ModuleNotFoundError: No module named 'klayout.tlcore'
```

Are there plans to fix this?

ThomasHilger, 2024-11-22:
I just had the same issue.
The reason for my case was, that I created a conda-package for linux, but installed it under windows.

klayoutmatthias, 2024-11-22:
I don't have conda, but I tried with the standard Python distribution on Windows 11. No issues.

The trouble you have usually originates from DLL dependency issues. Auditwheel should prevent that, so    I think there is a problem on your side. The problems are difficult to debug. ProcMon is my friend in these cases.