KLayout/klayout #1439: klayout crashed

paramsaini, 2023-07-31:
when i am trying to open klayout then some error found on screen. I will share the scrrenshot.
[image]

paramsaini, 2023-07-31:
can someone help me out with this error.

sebastian-goeldi, 2023-08-01:
You are using a klayout version that is almost 1.5 years old. I would highly suggest to update klayout to the latest version (0.28.10). As it is package manager related, I would guess either it's an incompatible package or incompatibility between the saltmine and your klayout, though I'd highly suspect the former.

paramsaini, 2023-08-01:
thanks for reply Sebastian. I already build klayout through git hub repository through ./build.sh after that when I tried to open it, it gives an error. I don't know what's going wrong.  when I am trying to find out the version with klayout --version again it shows this error" info: warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway. ABORTED (core dumped)".
I also tried with "XDG_SESSION_TYPE=x11 klayout"  but it doesn't work.

paramsaini, 2023-08-01:
is there any thing I forget to do after ./build.sh??

klayoutmatthias, 2023-08-01:
There is an official .deb package provided here: https://www.klayout.de/build.html. You do not need to build yourself.

Is that a plain Ubuntu 22 installation? I'm using it myself, so I can't confirm there is an issue.

You can try running "klayout -d 31" and share the log. It may show what's wrong. From the stack trace I think some script is executed which does something nasty. Did you already install some script or packages?

You can try to set "KLAYOUT_HOME" to something like `~/.klayout2` (some non-existing folder) so disable execution of scripts you may have stashed already. KLayout should start then. 

Matthias