KLayout/klayout #1441: Support for label position in Marker Database Browser

EugeneZelenko, 2023-07-31:
It'll be great to add support for label position as in GDS format ("n/e/s/w" and their combinations) in Marker Database Browser, so multiple labels could be shown for single violation without using `\\n` in label text.

EugeneZelenko, 2023-08-01:
Hi, Matthias!

Sure, `halign` and `valign` should be enough. But does `rdb` reader support them? I looked on `_test_extractor_impl` for `Text/DText` and it seems to read only text and transformations, but not alignments.

klayoutmatthias, 2023-08-01:
You're right. That's not supported currently.

Matthias

closed 2023-08-06 by klayoutmatthias

EugeneZelenko, 2023-08-06:
Hi, Matthias!

Thank you for such fast implementation!

nmz787-intel, 2023-09-15:
Is this related to text Labels not showing rotation? I see my labels rotated in Calibre, but not KLayout.

EugeneZelenko, 2023-09-16:
Not, it was about text alignment relative to reference point (above/below/center and left/right/middle). See changes that fixed issues, especially test case.