KLayout/klayout #1447: Built-in Fonts not showing correctly on CentOS 7

zjujjm, 2023-08-04:
>
[image]

Hello, everyone,

When I installed Klayout on CentOS7 by rpm file, all the fonts were not showing correctly as shown in the attached file. 

I checked the language LC_ALL=en_US.utf-8 and LANG=en_US.utf-8. Others are default settings.

Does anyone have an idea to fix this?

Thanks.

chkkbim, 2023-08-04:
I searched for all issues using keyword "font".
Maybe this solution can help :
https://github.com/KLayout/klayout/issues/1105#issuecomment-1169302522

klayoutmatthias, 2023-08-05:
Did you install font packages? 

I need to install at least these minimum packages on a fresh CentOS7 Docker image:

dejavu-sans-mono-fonts dejavu-sans-fonts dejavu-serif-fonts dejavu-fonts-common

Matthias

zjujjm, 2023-08-12:
Hi, Matthias,

Thank you very much for your reply. I have installed all packages you mentioned. But did not solve the problem. I have tried 3 servers I have here and all have the same issue. Quite weird. 

Best regards,