KLayout/klayout #1448: Editor mode not working macOS Ventura

4ndrebar, 2023-08-05:
Currently using macOS Ventura 13.5 on a MacBook Air M1. When drawing a path or a polygon in editor mode the white outline is shown but once double click to put down the cell it is not displayed in the selected layer.

klayoutmatthias, 2023-08-06:
Please paste a screen shot of what you are doing.

4ndrebar, 2023-08-06:
I tried on a Windows machine with the same mouse actions, and it works seamless.

https://github.com/KLayout/klayout/assets/47941691/3824cf4b-f046-4bb5-af55-dd3aa863a532

klayoutmatthias, 2023-08-06:
I don't see layer styles and colors. I suspect there is an issue with painting in general. There have been reports about problems with high-DPI displays, but I can't reproduce them myself as I'm not a Mac user and there have been some code changes to deal with that problem.

First of all, the version you're using is very old. Please try with the latest version and get back here if the problem can be reproduced there.

Matthias

4ndrebar, 2023-08-06:
Dear Matthias, thank you for addressing the issue. 
I updated to the 0.28.10, but the same issue is present.  
Now the colored box of the layer is displayed on the column on the right.
When double-clicking the outline disappears and no cell is displayed.

klayoutmatthias, 2023-08-06:
It is normal that the layer outline disppears when double-clicking. That is because double-clicking makes the layer hidden.

What happens if you load a layout file? Can you see some layout then?

Matthias

closed 2023-08-22 by 4ndrebar