KLayout/klayout #1449: Duplicate placements for GDSII but not OASIS files in  `KLayout.db.Layout.read()` when a cell >1 hierarchy levels below  is read twice

ejprinz, 2023-08-06:
# Bug Report for `KLayout.db.Layout.read`

There is an inconsistency in how layout is being read between
GDSII and OASIS files. The 4 layout reading options are:

- `AddToCell`, `OverwriteCell`, `RenameCell`, `SkipNewCell`

The expectation is that duplicate cells or shapes are not being
read into the layout and placed again. This works for _OASIS_ files.
However, for _GDSII_ files this is only observed for the top cell but not the
lower level cells. In other words, the cells and shapes under the top cell
are placed once, but the cells and shapes in lower level cells are placed
every time the cell is being read:

To show this I ran the following loop:

```python

    for layout_format in ['oas', 'gds', 'gds.gz']:
        for cell_conflict_resolution in ['AddToCell', 'OverwriteCell', 'RenameCell', 'SkipNewCell']:
            result = test_case(layout_format, cell_conflict_resolution)
            list_instances(result)

```

and the output is:
```sh

Cells and Instances for file tc_AddToCell.oas:
... tc_AddToCell ==> cell_index=1 r0 500,500
... hier1 ==> cell_index=0 r0 5,5

Cells and Instances for file tc_OverwriteCell.oas:
... tc_OverwriteCell ==> cell_index=1 r0 500,500
... hier1 ==> cell_index=0 r0 5,5

Cells and Instances for file tc_RenameCell.oas:
... tc_RenameCell ==> cell_index=1 r0 500,500
... hier1 ==> cell_index=0 r0 5,5

Cells and Instances for file tc_SkipNewCell.oas:
... tc_SkipNewCell ==> cell_index=1 r0 500,500
... hier1 ==> cell_index=0 r0 5,5

Cells and Instances for file tc_AddToCell.gds:
... tc_AddToCell ==> cell_index=1 r0 500,500
... hier1 ==> cell_index=0 r0 5,5
... hier1 ==> cell_index=0 r0 5,5                 <== only expect one instance, not two

Cells and Instances for file tc_OverwriteCell.gds:
... tc_OverwriteCell ==> cell_index=1 r0 500,500
... hier1 ==> cell_index=0 r0 5,5
... hier1 ==> cell_index=0 r0 5,5                 <== only expect one instance, not two

Cells and Instances for file tc_RenameCell.gds:
... tc_RenameCell ==> cell_index=1 r0 500,500
... hier1 ==> cell_index=0 r0 5,5
... hier1 ==> cell_index=0 r0 5,5                 <== only expect one instance, not two

Cells and Instances for file tc_SkipNewCell.gds:
... tc_SkipNewCell ==> cell_index=1 r0 500,500
... hier1 ==> cell_index=0 r0 5,5
... hier1 ==> cell_index=0 r0 5,5                 <== only expect one instance, not two

Cells and Instances for file tc_AddToCell.gds.gz:
... tc_AddToCell ==> cell_index=1 r0 500,500
... hier1 ==> cell_index=0 r0 5,5
... hier1 ==> cell_index=0 r0 5,5                 <== only expect one instance, not two

Cells and Instances for file tc_OverwriteCell.gds.gz:
... tc_OverwriteCell ==> cell_index=1 r0 500,500
... hier1 ==> cell_index=0 r0 5,5
... hier1 ==> cell_index=0 r0 5,5                 <== only expect one instance, not two

Cells and Instances for file tc_RenameCell.gds.gz:
... tc_RenameCell ==> cell_index=1 r0 500,500
... hier1 ==> cell_index=0 r0 5,5
... hier1 ==> cell_index=0 r0 5,5                 <== only expect one instance, not two

Cells and Instances for file tc_SkipNewCell.gds.gz:
... tc_SkipNewCell ==> cell_index=1 r0 500,500
... hier1 ==> cell_index=0 r0 5,5
... hier1 ==> cell_index=0 r0 5,5                 <== only expect one instance, not two

```

I think the expectation is that cells and shapes are not being placed twice if a cell is
read in twice, but only once.
Note that I had to rename the Python file as `.py.txt` to be able to upload it - Rename it to `.py` to run it.

[out.txt](https://github.com/KLayout/klayout/files/12268561/out.txt)
[README.md](https://github.com/KLayout/klayout/files/12268562/README.md)
[test_case.py.txt](https://github.com/KLayout/klayout/files/12268566/test_case.py.txt)

klayoutmatthias, 2023-08-06:
Hello @ejprinz,

thanks a lot for this perfectly prepared testcase!

But I'm afraid the problem is somewhere else: The property to set the cell conflict resolution mode is lower case "cell_conflict_resolution_mode". So instead of 

```
        reader_options.CellConflictResolution = reader_options.AddToCell
```

it should be 

```
        reader_options.cell_conflict_resolution = reader_options.AddToCell
```

With this modification I see the effect that I'm expecting.

The OASIS *writer* inherently does duplicate object removal. That is a consequence of the attempt to create object arrays (there is not array with zero displacement), so in the OASIS case duplicates are removed even if they are first generated by "AddToCell" mode.

Matthias

ejprinz, 2023-08-06:
Hallo Matthias:

Thanks for the fast response, I confirmed that with this change it works.
Definitely worth writing the test case !! It's interesting that Python
allows me to write this code, I need to find out which IDE tool/extension
in Neovim could have shown me the issue.

Best regards, Erwin

klayoutmatthias, 2023-08-06:
You're welcome :)

In Python, classes are not "closed" and Python allows creating new attributes on objects with arbitrary names. This is what happened in your case. I do not know if there is a way to raise an error if a property name is incorrect - there are schemes that override `__setattr__` to catch such issues, but I have not tried any of them.

Matthias

sebastian-goeldi, 2023-08-07:
For neovim, I think any standard language server will catch this (see for a start https://github.com/neovim/nvim-lspconfig#quickstart ). I can personnaly recoomend ruff-lsp if you want a fast language server or python-lsp-server if you want a more complete one (also has handy plugins for static type checker like pylsp-mypy).

Wrt to `__setattr__` there are ways to do it. E.g. pydantic does that (you'll get a validation error unless a specific flag is set). But in general I don't think this is a wise decision unless it is for a specific purpose such as the pydantic case. It's not cheap computationally and I don't see a good reason to do it, as this is a problem with pretty much any python class. A good lsp or even some linters will catch this. Also, a plea from me, don't implement any of those checks, there's a good chance you'd break a lot of my code ^^

klayoutmatthias, 2023-08-08:
Understood .. I do not plan to add more than necessary :)

But I myself came across the problem of mistyped properties. With the Python stubs PyCharm for example should be able to find these mistakes too.

Can I close this ticket?

Thanks,

Matthias

ejprinz, 2023-08-08:
Yes, you can close this ticket. Thanks.

closed 2023-08-09 by klayoutmatthias