KLayout/klayout #1452: compiling with qt6 shows one error, can't quit klayout on MacBook M1

stefanottili, 2023-08-06:
MacBook Air M1, MacOS Ventura

brew uninstall qt5
brew install qt6
build4mac.py -r HB31 -p HBAuto -q Qt6Brew -m '--jobs=8' -n -u

../../../src/layui/layui/layWidgets.cc:1093:11: error: no member named 'setMargin' in 'QHBoxLayout'
  layout->setMargin (0);
  ~~~~~~  ^
Commenting out line 1093 and one can build a working application.

1) Open gds/oasis, view layout, run xor, all works. But one can't quit the app.

klayout->Quit lvs_install.lym doesn't work, neither does hitting the red "close window" button on a layout window.

I had to use "kill -9 <pid>" or "Activity Monitor->force quit" to close klayout.

2) About and Quit display lvs_install.lym string in the menu (independent of qt5 or qt6)

klayout->About lvs_install.lym and klayout->Quit lvs_install.lym
Shouldn't this be klayout->About klayout and klayout->Quit ?

klayoutmatthias, 2023-08-07:
The build issue should be fixed in master, but the menu item display issue is weird. "About" and "Quit" should never be modified. I wonder if that is a Qt 6 issue or there is some problem with the way the menus are built on Ventura under Qt.

stefanottili, 2023-08-07:
2) klayout menu on Mac 

[image]

qt5/qt6 behave the same way on a MacBook Air M1 MacOS Ventura.

Kazzz-S, 2023-08-07:
On an Intel Mac (Monterey + Homebrew), I built KLayout with the same patch above.

```
./build4mac.py -r HB31 -p HBAuto -q Qt6Brew -m '--jobs=4' -n -u
```

In my environment, the menu looks normal.
[image]
[image]

However, `Quit klayout` did not work, and I also had to use "kill -9 <pid>" or "Activity Monitor->force quit" to close the application.

klayoutmatthias, 2023-08-08:
Honestly, I have no idea where this menu problem comes from neither why Quit won't exit.

There is nothing special with these menus and KLayout adding something to the menus should happen only for certain menus such as the DRC and LVS dock menus below the Tools/DRC and Tools/LVS menus. 

I understand that the MacOS menu system is very different from the traditional menus on Windows and Linux, so maybe special care needs to be taken there. But that is the reason we use frameworks - to take away the burden of handling OS specific details.

I just released 0.28.11 - do the Qt5 builds there show similar issues too?

Matthias

stefanottili, 2023-08-09:
menu issue shows only on M1/Ventura, both qt5/qt6. no menu issue on x86/Monterey either qt5 or qt6.
klayout can't quit both on x86/Monterey and M1/Ventura qt6, qt5 is fine on both.

stefanottili, 2024-02-01:
Superseded by https://github.com/KLayout/klayout/issues/1602

closed 2024-02-01 by stefanottili