KLayout/klayout #1457: klayout import def warns about "Cannot determine routing layers for via"

stefanottili, 2023-08-10:
Warning: Cannot determine routing layers for via: turnmet3 (line=5, cell=, file=turn_via.def)
Warning: Cannot determine routing layers for via: via23 (line=9, cell=, file=turn_via.def)
Warning: Cannot determine routing layers for via: via12 (line=13, cell=, file=turn_via.def)

The lef contains TYPE CUT/TYPE ROUTING for the via/metal layers.
The import look ok because there are no path segments following the via references anyways.
So why the warnings ?

The order of the layers in a via definition is not guaranteed, so should be derived from the lef layer definitions.

[test_turn_via.tgz](https://github.com/KLayout/klayout/files/12308060/test_turn_via.tgz)

klayoutmatthias, 2023-09-13:
I missed that on 0.28.12. Will try to do it in the next minor release.

klayoutmatthias, 2023-10-06:
I only get this warning if the LEF is not seen.

In the "strm2oas --lefdef-lefs sky130_fd_sc_hd.tlef turn_via.def a.oas" the LEF is explicitly given. When I load with KLayout, by default all LEFs beside the DEF are read, but KLayout checks for "*.lef", not "*.tlef" and this warning happens. It disappears when I add the *.tlef file to the reader options. 

Is "tlef" a common extension for technology LEF? In that case I'd just add "tlef" as the recognized LEF extensions.

Matthias

stefanottili, 2023-10-06:
It seems that OpenLane with it’s sky130 process uses tlef as extension for their technology lef

I’ve seen .lef_5.6, .lef_5.7, .lef_5.8 in the wild ….

And while we’re at this topic …. “don’t shoot the messenger”:

Si2 has ratified lef 6.0 and that changes (aka “make better”) the PINS placement, among some other syntax changes.

stefanottili, 2023-10-13:
https://github.com/KLayout/klayout/pull/1504

fixes this issue too.
It was caused by the specified .tlef not being read at all.
It would still be nice to have .tlef in the list of lef file extensions.

closed 2023-11-07 by stefanottili