KLayout/klayout #1459: Can I use the editing function of the kweb project？

huangwenjunlovedy, 2023-08-14:
Dear author, I have used the functionality of kweb in my own project, but kweb cannot move and edit gds. What should I do?
[image]

I hope to have MOVE functionality like the Editor version of Klayout. thanks
[image]

klayoutmatthias, 2023-08-14:
Yes, that is basically possible, although limited to very basic editing.

You need to create an editable LayoutView on the server side, like this:

```
pya.LayoutView(true)
```

The demo code will then allow selecting shapes and moving them. Also new modes should become available such as "polygon", "box" etc.

Please make sure you use the latest klayout module as previous versions had some bugs in that area.

Matthias

sebastian-goeldi, 2023-09-06:
Btw, this should be the default behavior in the new kweb. Just be aware that a lot of the edit functionalities don't work, because double click isn't handled properly (yet). Additionally, layer selection also doesn't work yet, therefore polygon drawing etc, doesn't male much sense yet. This is pretty low on the priority list for me, as I don't see huge use cases for any of those functionalities.

Anyway, I think this is much more a [kweb](https://github.com/gdsfactory/kweb) problem than klayout. If you have further problems with kweb, I would encourage you to open issues there, instead of directly here, as Matthias is involved in kweb by association because he wrote the prototype for it, canvas2canvas. But most of the current crimes in kweb, are probably mine ;)

klayoutmatthias, 2023-09-07:
@sebastian-goeldi  Very good. Thanks for commenting :)

You're right, a lot of UI wiring needs to be done and for sure there is a limit somewhere. After all, rebuilding the entire functionality of the client application is not really desirable.

Matthias