KLayout/klayout #1462: Ubuntu download does not contain build.sh

joemos11, 2023-08-14:
From the page
_https://www.klayout.de/build.html_
downloaded the following file for an Ubuntu 22 installation
klayout_0.28.11-1_amd64.deb
Once in home directory ran
ar x klayout_0.28.11-1_amd64.deb
to extract the .deb and then
tar hzxvf data,tar.gz
to extract the data file, which makes a ,/usr directory 
However, when I follow the instructions and 
cd ./usr
./build.sh -python
It cannot find the build.sh script.
I thought maybe this was a completely built klayout, so I then tried
cd ./bin
klayout
However it said that klayout was not found. The command line suggested running
sudo apt install klayout
but the version that it would load is 0.26.2-3build3

Help!

chkkbim, 2023-08-15:
If you want to install and run, you should use the deb file.
You can follow these general instructions :
[https://phoenixnap.com/kb/install-deb-files-ubuntu](url)

If you want to compile and build from scratch, you have to download the source files.
Then compile and build inside a dockerfile :
[https://github.com/KLayout/klayout/issues/1105#issuecomment-1169302522](url)

joemos11, 2023-08-15:
Many thanks. You may want to consider updating the installation instructions to include the link to how to install .deb packages.

closed 2023-08-15 by joemos11

sebastian-goeldi, 2023-08-15:
I don't think that's a problem of klayout. A .deb is the standard way to install packages properly (except for snap & flatpak or AppImages) on any debian or debian derivative. If you are on ubuntu, a double click on the deb file should even take you to the GUI installer afaik.

The build site probably doesn't tell you how to install it because then it would have to do the same for the other 5 or so OS families. And that's simply asking quite a bit, also to keep it up to date.

For reference, in the future, you can either install it by GUI or in a terminal with `sudo apt install ./klayout-[...].deb`. Note that there must be a `./` or another path specifictation as otherwise the package manager will default to the apt repositories for loookup & installation, which is why you got 0.26.x, most likely because you are on an older ubuntu version or because a newer version of klayout wasn't uploaded to the ubuntu default repositories.

Also theoretically you can install it with the low-level utility `sudo dpkg -i klayout-[...].deb`  but this won't install/check dependencies.