KLayout/klayout #1463: feature request : add "TypeArray"(TypeList) to pcell varible type

dsus4wang, 2023-08-16:
Hi,
Sometimes when I write a pcell (using ruby) , I need to parse a Array to the pcell.
Now I define the "array" varible using "TypeString" , When I draw layout by code ,it works.
But when I use GUI (double click pcell) to modify the varibles of pell , It will print error (the array will turn to string)
I tried to rewrite the String class
```
class String
	def each()
		p self
		eval("[#{self}]").each
	end
end
``` 
When I retry to modify the pcell , klayout won't return error ,  but i found the pcell varible still was type string..It doesn't work
I think maybe we can add a new type to RBA class?
thanks

klayoutmatthias, 2023-08-19:
I'm not sure I understand.
Could you paste some sample PCell code?

Matthias

klayoutmatthias, 2023-08-20:
I am still trying to understand your use case.

There is "TypeList" and you can use it in this way (Python):

```
    self.param("ll", self.TypeList, "A list", default = [])
```

However, on the user interface side, that is only a test edit box that accepts a comma-separated list of values which are turned into strings - i.e. input "1,2" will give you "[ '1', '2' ]".

That is a little simplistic, but that is the way it is currently.

You can always turn a list of strings into floats for example with such a piece of code (Python, assuming the list parameter is "ll"):

```
    ll_values = [ float(v) for v in self.ll ]
```

If you put this code into "coerce_parameters" too (the validation callback), KLayout will reject invalid entries, like here:

[image]

Right now, the list type cannot be specified further in terms of the element type and there is no specific user interface provided for the list.

Matthias

dsus4wang, 2023-08-22:
Ah that is what I need

dsus4wang, 2023-08-22:
Thank you Matthias!

closed 2023-08-22 by dsus4wang