KLayout/klayout #1465: DRC Internal Error after Completion

yrrapt, 2023-08-23:
At the end of running a DRC script there is an error:

```
Total elapsed: 190.680s  Memory: 4192.00M
ERROR: RuntimeError: Internal error: ../../../src/db/db/dbLayout.cc:2035 m_layers.layer_state (n) != LayoutLayers::Free was not true in Layout::delete_layer in Executable::cleanup
  :/built-in-macros/_drc_source.rb:95:in `delete_layer'
  :/built-in-macros/_drc_source.rb:95:in `block in finish'
  :/built-in-macros/_drc_source.rb:94:in `each'
  :/built-in-macros/_drc_source.rb:94:in `finish'
  :/built-in-macros/_drc_engine.rb:2711:in `block in _flush'
  :/built-in-macros/_drc_engine.rb:2710:in `each'
  :/built-in-macros/_drc_engine.rb:2710:in `_flush'
  :/built-in-macros/_drc_engine.rb:2720:in `_finish'
  :/built-in-macros/drc_interpreters.lym:47:in `cleanup'
```

This is found version 28.11 but running the same DRC script on another computer with KLayout 28.6 does not have the error.

Thanks,
Thomas

klayoutmatthias, 2023-08-24:
Hi Thomas,

thanks for reporting that issue. 

I think I understand the execution path that leads to this issue, but I can't reproduce the error. I think it has to do with source manipulations ("clip", "cell", "select" etc.) and "extent" with a cell filter (i.e. "extent('*')").

Can you confirm that?

Matthias

klayoutmatthias, 2023-09-02:
Commit 304800e4c59e0238ead91ae814093b22b6545cc3 may fix this issue. This change will go into the next minor version. Let's see if the problem persists.

closed 2023-09-13 by klayoutmatthias

yrrapt, 2023-09-16:
Sorry for the delayed response, but I can now confirm that the issue is fixed with the latest release.
Thank you!