KLayout/klayout #1467: How to customize layer view's color and stipple style with python script?

drunkwcodes, 2023-09-01:
Hi, I'm trying to automate these setting processes by using klayout python script.

Please tell me how to set the layer color and stipple style. I want to make the layer a black & white image.

[image]

I used `pya` to do the klayout script before.

```
import pya

app = (
    pya.Application.instance()
)  # TODO: FIXME module 'pya' has no attribute 'Application'
mw = app.main_window()

current_vew = mw.current_view()
current_vew.min_hier_levels = 0
current_vew.max_hier_levels = 0
# save_image = current_vew.get_screenshot()
# save_image.save(box_path)
box_path = os.path.join(dir1, "bbox.png")
resolution = int(importECADDict["resolution"])
current_vew.save_image(box_path, resolution, resolution)
```

Please help me to figure out how to refine the script, thanks.

klayoutmatthias, 2023-09-02:
I think what you are looking for is this function: https://www.klayout.de/doc-qt5/code/class_LayoutView.html#method85

Inside the loop, you can customize layer view properties depending on the layer's "source" which gives you the layer and datatype number.

drunkwcodes, 2023-09-03:
Please give more detail. I can not see where to set view properties like "stipple" and "color".

And I'm using python, thanks.

s910324, 2023-09-06:
```python
import pya
layoutView = pya.Application.instance().main_window().current_view()  
for lyp in layoutView.each_layer():
    lyp.fill_color     = 255 
    lyp.frame_color    = 255
    lyp.dither_pattern = 0
```

drunkwcodes, 2023-09-06:
@s910324 Thank you! I will try it.

drunkwcodes, 2023-09-06:
@s910324 The `fill_color` settings in the script are not working. I don't know why.

Please help me to figure it out. thanks.
I provide the test project in my github: https://github.com/drunkwcodes/inp_editing

There are sample gds file "PNL.gds" and my script. The command is in the README.

s910324, 2023-09-06:
HI fill_color and frame_color take one int as parameter, 
you might need to convert hex or any given color format to int before passing the value

black#000000 => 0
blue #0000ff => 55
aqua #00ffff=> 65535
white #ffffff==> 16777215
https://www.klayout.de/doc-qt5/code/class_LayerProperties.html#method38

drunkwcodes, 2023-09-06:
@s910324  I got it.

The setting are effective.

But I saved the wrong image. `layout_view.save_image()` seems to save the unset layers to me.

How to change this? Using `layout_view.get_screenshot()`?

s910324, 2023-09-06:
you might need to hide layers before you capturing,  this shows 2/0, 3/0 and hide others

showlayers = [(2, 0), (3, 0)]
for lyp in layoutView.each_layer():
    lyp.visible        = (lyp.source_layer, lyp.source_datatype) in showlayers
    lyp.fill_color     = 255
    lyp.frame_color    = 255
    lyp.dither_pattern = 0

drunkwcodes, 2023-09-06:
@s910324 Yes I did hide unwanted layer.

I can see my ideal screenshot after running my macro.
But I just can not get that screenshot. Please help.
[image]

```python

"""Read gds file and export black white image."""

from pathlib import Path
import pya
from time import sleep

gds_path=str(r"D:\projects\inp_editing\PNL.gds")
output_path=str(r"D:\projects\inp_editing\bw3.png")
app = (
    pya.Application.instance()
)  # TODO: FIXME module 'pya' has no attribute 'Application'
mw = app.main_window()

mw.load_layout(gds_path, 2)
# mw.current_layout()

layout_view = mw.current_view()  
for lyp in layout_view.each_layer():
    lyp.fill_color     = 0
    lyp.frame_color    = 0
    lyp.dither_pattern = 0
    

    

    # # Set the visibility of the layer (True for visible, False for invisible)
    # layer_info.visible = False  # Make the layer invisible
    if lyp.layer_index() == 4:
        lyp.visible = False
sleep(1)
screenshot = layout_view.get_screenshot()
screenshot.save(output_path, "PNG", -1)

# layout_view.save_image(output_path, 2000, 2000)

```

s910324, 2023-09-06:
Hi there, 

I see what you mean by now, change the ```sleep(1)``` to ```pya.Application.instance().process_events()```

this should make the layer visibility change takes effect while the screen shot is taken

drunkwcodes, 2023-09-06:
@s910324 Thank a lot! It works!

But I have a question. It can only executable in GUI mode, which means that I can not get the screenshot when passing `-b` parameter in my command. Do you know why?

It didn't generate any error message in `stdout`.

s910324, 2023-09-06:
For headless image capture, there's not going to have a mainwindows involved, so everything goes into layout view directy.

this thould works.

```python
app = pya.Application.instance()
layout_view = pya.LayoutView()
layout_view.load_layout(gds_path)
layout_view.max_hier() 

for lyp in layout_view.each_layer():
    lyp.fill_color = 0
    lyp.frame_color = 0
    lyp.dither_pattern = 0
    if lyp.layer_index() in [4]:
        lyp.visible = False

layout_view.update_content()
screenshot = layout_view.get_image(1000, 1000)
screenshot.save(output_path, "PNG", -1)
```

drunkwcodes, 2023-09-06:
@s910324 Thanks for your quick response. I will try tomorrow.

drunkwcodes, 2023-09-07:
@s910324 The image became all black. I don't know why.

s910324, 2023-09-07:
@drunkwcodes 

If you recently changed your viewer to a dark background, the screen shot will becomes all black since the layers are all set to black.

to avoid this factor from user side you might need to set the view port color to desired bg-color before apply screen shot

```python
layout_view.set_config("background-color", "#ffffff")
layout_view.set_config("grid-show-ruler", "false")
layout_view.set_config("grid-visible", "false")
```
these three lines of code set bg-color to white and remove the ruler as well as grid from screen shots

hope this solves the issue

drunkwcodes, 2023-09-07:
@s910324 Thanks a lot. It works!

This is so important for my work that I even want to sponsor you!
If you open sponsor function in github, please notify me.

I'm following you!

closed 2023-09-07 by drunkwcodes

drunkwcodes, 2023-09-12:
@s910324 I have another question: how to pass variables from command line to klayout script?
"-rd" option seems not working in python scirpt.

s910324, 2023-09-12:
@drunkwcodes 

for commandline please try this:
```
 .\klayout_app.exe -b -r test.py -rd gds_path=.\PNL.gds -rd output_path=.\BK.png
```

For the .py part:
1. declear your parameters as a python global
2. remove parameter initiallization, such as `gds_path = ""` 

```python
import pya

global gds_path
global output_path

layout_view = pya.LayoutView()
layout_view.load_layout(gds_path)
layout_view.max_hier() 

layout_view.set_config("background-color", "#ffffff")
layout_view.set_config("grid-show-ruler", "false")
layout_view.set_config("grid-visible", "false")

for lyp in layout_view.each_layer():
    lyp.fill_color     = 0
    lyp.frame_color    = 0
    lyp.dither_pattern = 0
    
    if lyp.layer_index() in [4]:
        lyp.visible = False

layout_view.update_content()
screenshot = layout_view.get_image(1000, 1000)
screenshot.save(output_path, "PNG", - 1)
```

drunkwcodes, 2023-09-12:
Thanks a lot, I will try it tomorrow.

drunkwcodes, 2023-09-13:
@s910324 It works, thanks!

drunkwcodes, 2023-09-18:
Hi @s910324 ,

this line can not run on klayout 0.24.10

```
Invalid number of arguments (got 1, expected 2/3/4) in LayoutView.load_layout
  .\kl_script_gen_bw_image.py:24
```

Do you know how to call `load_layout` in old version?
There are missing args.

I also have some old script having issues when running on 0.28, so I decide to change my newer script(this script)
to run on old version, thanks.

s910324, 2023-09-19:
@drunkwcodes 

for 0.24.10, the document seems to be the same as current version
https://www.klayout.de/0.24/doc/code/class_LayoutView.html#method92

But I found LayoutView object is not as powerful as new version, and seems like cannot be created individually without the existenceof mainwindow, so code targeting 0.24.10 requires some change.

and this works under `.\klayout_app.exe -z -r` command, but not with `-b` command lie this `.\klayout_app.exe -b -r `
due to `-b` supress the mainwindow.

```python
main_window = pya.Application.instance().main_window()
main_window.load_layout(gds_path, True)

layout_view = main_window.current_view()
layout_view.max_hier() 
layout_view.set_config("background-color", "#ffffff")
layout_view.set_config("grid-show-ruler", "false")
layout_view.set_config("grid-visible", "false")

layer_iter = layout_view.begin_layers()
while not(layer_iter.at_end()):

    lyp                    = layer_iter.current()
    new_lyp                = lyp.dup()
    new_lyp.fill_color     = 0
    new_lyp.frame_color    = 0
    new_lyp.dither_pattern = 0
    
    if lyp.layer_index() in [4]:
        new_lyp.visible = False
    
    layout_view.set_layer_properties(layer_iter, new_lyp)
    layer_iter.next()
    
layout_view.update_content()
screenshot = layout_view.get_image(1000, 1000)
screenshot.save(output_path, "PNG", - 1)
```

drunkwcodes, 2023-09-19:
@s910324 It works again!

You're welcome for your tremendous help, which has allowed my work to thrive. The day of releasing the new version of the program is approaching. Having you is truly wonderful.

Your knowledge and experience are truly amazing, making me feel outmatched. I hope to have you by my side throughout my journey with KLayout, helping me overcome the challenges of the upcoming upgrades to 0.28.

Thanks and best regards,
Eric