KLayout/klayout #1472: strm2oas: def path with first/last segment length < halfwidth are read wrong

stefanottili, 2023-09-07:
strm2oas by default decomposes multi-point path with first or last segment length < halfwidth incorrectly.
They are read correctly as multi-point path by using the --lefdef-joined-paths option.

make bad; make xor   shows the difference
make good; make xor  doesn't

[test_spnet.tgz](https://github.com/KLayout/klayout/files/12547728/test_spnet.tgz)

klayoutmatthias, 2023-09-07:
Thanks! Perfect testcase preparation!

But hard to fix :(

klayoutmatthias, 2023-09-07:
I mean, I don't know if the "good" version is even correct - that assumes a "GDS style" interpretation of path. Can you confirm the way these paths are build is compatible with other tools?

Like this "good" path:

[image]

stefanottili, 2023-09-07:
It certainly looks like unintended,  but was found in the wild.I would say the fix is to warn about it and not break it into two point path.The multipoint path is handled correctly after all.I have an option to turn this into a polygon, because somebody got burned somewhere years ago by some tool not doing the right thing ….

klayoutmatthias, 2023-09-09:
Okay ... my fix now is to join the first two segments if the initial segment is too short. Same for the last two segments. 

Matthias

stefanottili, 2023-09-09:
more elaborate then what I would have done, but I'll take it.
I would be nice to get a warning about this situation.

closed 2023-09-12 by klayoutmatthias