KLayout/klayout #1481: Klayout quits unexpectedly after building from source code on MacBook Air M1, MacOS Ventura 13.5

jasminabrar, 2023-09-11:
I have tried building KLayout from source code on my MacBook Air M1, MacOS Ventura using the instructions under 6B-6D in the official documentation (https://github.com/lukasc-ubc/klayout/tree/master/macbuild) as well as following the command from issue #1452. Using the command from issue #1452 I have attempted several combinations of ruby and qt versions (ruby3.1, ruby3.2, qt5, qt6), as I suspected the errors presented during KLayout's crash may have been caused by building with a wrong software version. However, regardless of the build method I use, KLayout quits unexpectedly when I attempt to open the application due to an invalid code signature. 

I am unsure of how to proceed and am reaching out to the community for help on how I should proceed. Some individuals I have seem in related GitHub issues are @klayoutmatthias @Kazzz-S @stefanottili @thomaslima. Any of your input would be greatly appreciated. Thank you. 

Here is the content of the error report:

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Incident Identifier: CF4BF077-5508-45F9-9BAF-A6169832EC69
CrashReporter Key:   612438D0-5FCF-B469-5D0D-17FBD3BA95A2
Hardware Model:      MacBookAir10,1
Process:             klayout [41358]
Path:                /Applications/klayout_githubissues_ruby3.1.app/Contents/MacOS/klayout
Identifier:          de.klayout
Version:             0.28.11 (0.28.11)
Code Type:           ARM-64 (Native)
Role:                Background
Parent Process:      launchd [1]
Coalition:           de.klayout [86947]

Date/Time:           2023-09-11 14:02:13.6606 -0700
Launch Time:         2023-09-11 14:02:13.1950 -0700
OS Version:          macOS 13.5 (22G74)
Release Type:        User
Report Version:      104

Exception Type:  EXC_BAD_ACCESS (SIGKILL (Code Signature Invalid))
Exception Subtype: UNKNOWN_0x32 at 0x000000010265c000
Exception Codes: 0x0000000000000032, 0x000000010265c000
VM Region Info: 0x10265c000 is in 0x10265c000-0x10266c000;  bytes after start: 0  bytes before end: 65535
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  __TEXT                      10265c000-10266c000    [   64K] r-x/r-x SM=COW  
      __DATA_CONST                10266c000-102670000    [   16K] rw-/rw- SM=COW  
Termination Reason: CODESIGNING 2 Invalid Page

Triggered by Thread:  0

Thread 0 Crashed:
0                                 	       0x102a161e4 dyld3::MachOFile::forEachLoadCommand(Diagnostics&, void (load_command const*, bool&) block_pointer) const + 52
1                                 	       0x102a181cc dyld3::MachOFile::forEachSupportedPlatform(void (dyld3::Platform, unsigned int, unsigned int) block_pointer) const + 160
2                                 	       0x102a69e0c dyld3::MachOFile::isBuiltForSimulator() const + 124
3                                 	       0x102a198e4 start + 632

Thread 0 crashed with ARM Thread State (64-bit):
    x0: 0x000000010265c000   x1: 0x000000016d7a3828   x2: 0x000000016d7a37d0   x3: 0x0000000102a15e2e
    x4: 0x0000000000000075   x5: 0x0000000000000075   x6: 0x0000000000000000   x7: 0x0000000000000a40
    x8: 0x000000016d7a3808   x9: 0x0000000102aa7530  x10: 0x0000000102a92000  x11: 0x0000000102a9e9b9
   x12: 0x0000000000000065  x13: 0x0000000000000073  x14: 0x0000000000056250  x15: 0x0000000000000000
   x16: 0x0000000102a1826c  x17: 0x6ae100016d7a37d0  x18: 0x0000000000000000  x19: 0x000000016d7a3828
   x20: 0x000000010265c000  x21: 0x000000016d7a37d0  x22: 0x000000016d7a3a08  x23: 0x000000016d7a39e0
   x24: 0x000000016d7a3960  x25: 0x0000000000000000  x26: 0x0000000000000000  x27: 0x0000000000000000
   x28: 0x0000000000000000   fp: 0x000000016d7a37c0   lr: 0x190b800102a181cc
    sp: 0x000000016d7a3730   pc: 0x0000000102a161e4 cpsr: 0x80001000
   far: 0x000000010265c000  esr: 0x92000007 (Data Abort) byte read Translation fault

Binary Images:
       0x102a14000 -        0x102aa3fff  (*) <e7a99595-e0f8-34af-be8b-9347d0d658a4> ???
       0x10265c000 -        0x10266bfff  (*) <05fef845-122a-3dfb-a85c-d8c319218603> ???

Error Formulating Crash Report:
dyld_process_snapshot_get_shared_cache failed

EOF

-----------
Full Report
-----------
[... 94 lines cut ...]
      "deploymentId" : 900000017
    },
    {
      "treatmentId" : "3a3cf641-8471-4e4e-9ad4-81d0ede970fd",
      "experimentId" : "64a84dae90d82611a0bd7d3d",
      "deploymentId" : 400000010
    }
  ]
},
  "reportNotes" : [
  "dyld_process_snapshot_get_shared_cache failed"
]
}

Model: MacBookAir10,1, BootROM 8422.141.2, proc 8:4:4 processors, 8 GB, SMC 
Graphics: Apple M1, Apple M1, Built-In
Display: Color LCD, 2560 x 1600 Retina, Main, MirrorOff, Online
Memory Module: LPDDR4, Hynix
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4378), wl0: May 13 2023 07:20:48 version 18.20.383.15.7.8.150 FWID 01-b37727a5
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: MacBook Air, Apple Inc.
Thunderbolt Bus: MacBook Air, Apple Inc.

stefanottili, 2023-09-11:
I've never seen the issue you're facing.

I'm using homebrew to get the dependencies.
Running the latest MacOS Ventura 13.5.2, but that shouldn't matter:
I've been building klayout on my MacBook M1 since it came out.

build4mac.py -r HB32 -p HBAuto -q Qt5Brew -m '--jobs=8' -n -u

worked for me just a couple minutes ago and it run's without issues.
I don't need the bindings, so I use -n -u.

klayoutmatthias, 2023-09-11:
A quick search gives me this: https://developer.apple.com/forums/thread/130313

But I can't help further. MacOS is not my home turf.

Matthias

Kazzz-S, 2023-09-12:
I have extended the build system for Mac in response to someone's request to build KLayout on Apple silicon.
However, I do not have an M1 Mac and have not tested it by myself. My apologies :-(

As @stefanottili attempted above, M1 Mac users are supposed to use **Homebrew** installed in the default directory below.

```
# File: "macbuild/build4mac_env.py"
:
:
if Machine == "arm64": # Apple Silicon!
  DefaultHomebrewRoot = '/opt/homebrew'
else:
  DefaultHomebrewRoot = '/usr/local'
:
:
```

Kazzz-S

jasminabrar, 2023-09-12:
Thank you everyone for your suggestions. It is strange that building isn't working for me but is working on other Mac M1 Ventura. I am currently exploring issue 50 and will report back with my findings.

jasminabrar, 2023-09-15:
@stefanottili What version of KLayout are you building? Lukas is also having issues with KLayout crashing when building the 0.28.12, however he can successfully build 0.28.5.

stefanottili, 2023-09-15:
I'm always building master, which as of Sept14 is at the 0.28.12 tag.
And I've never encountered the crashes Lukas is seeing since I got my 2020 M1 MacBook (16GB).

dsus4wang, 2023-09-22:
just use "sudo" to build

thomaslima, 2023-11-13:
For the love of me I am not able to build a HW version of the klayout app on my M2 Mac (Sonoma). The best I was able to do was a LW version, which requires all brew dependencies to be installed (libgit2, qt@5, python@3.11 etc.). That's what I am going to equip on a personal level.

The macdeployqt tool is not able to properly codesign a version of the app that can pass the Gatekeeper (even with no python and no ruby). Given that the error doesn't occur with the LW version, the culprit must be in the mac deployment step of qt. I will try again once the migration to Qt6 is complete.

If anyone is curious I did the following:

- `brew install ccache qt@5 python@3.11 libgit2 libpng` (I might have forgotten something here)
- cloned klayout (I'm on kazzz/wip-mac-master's branch, commit SHA f57d2cbf5cf5ddae9fc2047b916e51926f0b955f)
- build with `QMAKE_CCACHE=1 ./build4mac.py -m --jobs=8 -r nil` (the ccache speeds up rebuilds in the future)
- make LW app with `QMAKE_CCACHE=1 ./build4mac.py -m --jobs=8 -r nil -Y` (I'm not sure if my edit below is required for the LW version)
- copy `LW-qt5Brew.pkg.macos-Sonoma-release-RnilPhb311/klayout.app` to my `/Applications` folder
- This will use python 3.11 installed in `/opt/homebrew/Cellar/python@3.11`

@Kazzz-S  I found that the `QMAKE_APPLE_DEVICE_ARCHS=arm64` environment variable is required for the Apple Silicon (It's possible Qt6 will have fixed that though). I have added the following lines to your script:

```
@@ -133,7 +134,8 @@ def Get_Default_Config():
         if Machine == "arm64" and Platform in ["Sonoma", "Ventura", "Monterey"]: # with an Apple Silicon Chip
             print("")
             print( "### Your Mac equips an Apple Silicon Chip ###" )
-            print("")
+            print("Setting QMAKE_APPLE_DEVICE_ARCHS=arm64\n")
+            os.environ['QMAKE_APPLE_DEVICE_ARCHS'] = 'arm64'
```

Kazzz-S, 2023-11-13:
@thomaslima Thanks for testing my `wip-mac-master` branch. Also, the change requested above.
I'll add the lines to the `build4mac.py` script.

Kazzz-S