KLayout/klayout #1483: Cannot make HW-klayout-0.28.12-macOS-Monterey-1-qt5Brew-RsysPhb39.dmg

Kazzz-S, 2023-09-14:
With the recent upgrade of Homebrew (early September 2023), I cannot correctly create **HW-klayout-0.28.12-macOS-Monterey-1-qt5Brew-RsysPhb39.dmg**. The build itself was successful. However, deployment went wrong. More precisely,

1. Qt5 binaries are not properly deployed. For example, `QtOpenGL.framework` is missing in the Frameworks directory.
2. Python3.9's dependencies, such as `mpdecimal, openssl@3, deadline,  sqlite, and xz`, are also missing in the Frameworks directory.

Kazzz-S, 2023-09-14:
## 1. Qt5 deployment ##

[This ticket](https://github.com/Homebrew/homebrew-core/issues/140930) captures the issue precisely and [this comment](https://github.com/Homebrew/homebrew-core/issues/140930#issuecomment-1703938107) provides a solution.
That is, by reinstalling locally, e.g. `brew update && brew reinstall qt@5 glib freetype`, the first issue has been fixed.

Kazzz-S, 2023-09-14:
## 2. Embedding Python3.9 ##

-- studying --

klayoutmatthias, 2023-09-16:
@kazzz Many thanks for your continued support! Please take your time.

Kazzz-S, 2023-09-23:
## 3. Downgrade Homebrew Python 3.9.18 ---> Python 3.9.17 (Part-I) ##

**Steps followed:**
1. Review the recent commit history of the python@3.9 formula.
           https://github.com/Homebrew/homebrew-core/commits/master/Formula/python@3.9.rb 
[image]

2. **git clone** https://github.com/Homebrew/homebrew-core.git and check out SHA1=**ebf6e554104d23d25f647d1fbac6fc7cf3a940a2**
[image]

3. Review the working directory.
```
(base) Ventura2{kazzz-s} homebrew-core (1)% pwd
/Users/kazzz-s/GitWork/homebrew-core

(base) Ventura2{kazzz-s} homebrew-core (2)% ll
total 60
drwxr-xr-x   17 kazzz-s staff    544  9 22 19:04 .
drwxr-xr-x   43 kazzz-s staff   1376  9 22 18:59 ..
drwxr-xr-x   14 kazzz-s staff    448  9 22 22:33 .git
drwxr-xr-x    7 kazzz-s staff    224  9 22 19:04 .github
drwxr-xr-x  300 kazzz-s staff   9600  9 22 19:04 Aliases
-rw-r--r--    1 kazzz-s staff   1459  9 22 19:03 CODEOWNERS
-rw-r--r--    1 kazzz-s staff   5525  9 22 19:04 CONTRIBUTING.md
drwxr-xr-x 6716 kazzz-s staff 214912  9 22 19:04 Formula
-rw-r--r--    1 kazzz-s staff   1334  9 22 19:03 LICENSE.txt
-rw-r--r--    1 kazzz-s staff    479  9 22 19:03 README.md
drwxr-xr-x   18 kazzz-s staff    576  9 22 19:04 audit_exceptions
drwxr-xr-x    5 kazzz-s staff    160  9 22 19:03 cmd
-rw-r--r--    1 kazzz-s staff   5046  9 22 19:04 formula_renames.json
-rw-r--r--    1 kazzz-s staff  22576  9 22 19:04 pypi_formula_mappings.json
drwxr-xr-x    9 kazzz-s staff    288  9 22 19:04 style_exceptions
-rw-r--r--    1 kazzz-s staff   2020  9 22 19:04 synced_versions_formulae.json
-rw-r--r--    1 kazzz-s staff   1144  9 22 19:03 tap_migrations.json

(base) Ventura2{kazzz-s} homebrew-core (3)% git status
On branch python-3.9.17 <=== OK
nothing to commit, working tree clean

(base) Ventura2{kazzz-s} homebrew-core (4)% git log --oneline
ebf6e554104 (HEAD -> python-3.9.17) python@3.9: update 3.9.17_1 bottle. <=== OK
2f4718afedc python@3.9: switch to `openssl@3`
18066a869e4 Merge pull request #134234 from carlocab/openssl-3-link
ea414a260c4 openssl@1.1: make keg_only on Linux
501a8a714db openssl@3: link into `HOMEBREW_PREFIX`
:
:
:
```

4. Unlink the python@3.9.18
```
(base) Ventura2{kazzz-s} homebrew-core (1)% pwd
/Users/kazzz-s/GitWork/homebrew-core

(base) Ventura2{kazzz-s} homebrew-core (2)% ll
total 60
drwxr-xr-x   17 kazzz-s staff    544  9 22 19:04 .
drwxr-xr-x   43 kazzz-s staff   1376  9 22 18:59 ..
drwxr-xr-x   14 kazzz-s staff    448  9 22 22:33 .git
drwxr-xr-x    7 kazzz-s staff    224  9 22 19:04 .github
drwxr-xr-x  300 kazzz-s staff   9600  9 22 19:04 Aliases
-rw-r--r--    1 kazzz-s staff   1459  9 22 19:03 CODEOWNERS
-rw-r--r--    1 kazzz-s staff   5525  9 22 19:04 CONTRIBUTING.md
drwxr-xr-x 6716 kazzz-s staff 214912  9 22 19:04 Formula
-rw-r--r--    1 kazzz-s staff   1334  9 22 19:03 LICENSE.txt
-rw-r--r--    1 kazzz-s staff    479  9 22 19:03 README.md
drwxr-xr-x   18 kazzz-s staff    576  9 22 19:04 audit_exceptions
drwxr-xr-x    5 kazzz-s staff    160  9 22 19:03 cmd
-rw-r--r--    1 kazzz-s staff   5046  9 22 19:04 formula_renames.json
[... 106 lines cut ...]
   [-u|-unlink]           : unlink only                       | disabled
   [-?|--?]               : print this usage and exit         | disabled
--------------------------------------------------------------+----------------------

Ventura2{kazzz-s} macbuild (4)% ./python3HB.py -v 3.9
```

8. Review the python@3.9.17_1 installation(2)
```
Ventura2{kazzz-s} ~ (1)% pwd
/Users/kazzz-s

Ventura2{kazzz-s} ~ (2)% which python3
/usr/local/opt/python@3.9/bin/python3 <=== OK

Ventura2{kazzz-s} ~ (3)% python3
Python 3.9.17 (main, Jun 20 2023, 17:20:08) <=== OK
[Clang 14.0.3 (clang-1403.0.22.14.1)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import pandas
>>> import scipy
>>> import matplotlib
>>> import klayout
>>> quit()
```

Kazzz-S, 2023-09-23:
## 4. Downgrade Homebrew Python 3.9.18 ---> Python 3.9.17 (Part-II) ##

**cont.**

9. Make the HW-qt5Brew.pkg.macos-Ventura-release-RsysPhb39 application bundle
            Note that **_qt5Brew.bin.macos-Ventura-release-RsysPhb39_** was built before upgrading Ventura to 13.6 (with the latest Xcode 15.0).
            Therefore, there was no build issue: https://www.klayout.de/forum/discussion/2377/ 
```
Ventura2{kazzz-s} klayout (1)% pwd

/Users/kazzz-s/GitWork/klayout
Ventura2{kazzz-s} klayout (2)% ll -rt
total 7388
:
:
drwxr-xr-x 210 kazzz-s staff    6720  9 19 18:33 qt5Brew.build.macos-Ventura-release-RsysPhb39
drwxr-xr-x 149 kazzz-s staff    4768  9 19 18:33 qt5Brew.bin.macos-Ventura-release-RsysPhb39
-rw-r--r--   1 kazzz-s staff 7141612  9 19 18:33 qt5Brew.build.macos-Ventura-release-RsysPhb39.log
drwxr-xr-x 212 kazzz-s staff    6784  9 19 18:34 qt5Brew.build.macos-Ventura-release-RsysPhb39.macQAT
drwxrwxr-x   4 kazzz-s staff     128  9 21 14:13 .vscode
-rw-r--r--   1 kazzz-s staff    6148  9 21 18:33 .DS_Store
drwxr-xr-x  16 kazzz-s staff     512  9 22 21:51 macbuild
drwxr-xr-x  17 kazzz-s staff     544  9 23 10:44 .git
drwxr-xr-x   6 kazzz-s staff     192  9 23 10:46 ..
drwxr-xr-x  44 kazzz-s staff    1408  9 23 11:09 .

Ventura2{kazzz-s} klayout (3)% ./build4mac.py -r sys -p hb39 -y

### You are going to make a heavyweight (HW-) package including Qt[5|6] and Python3.8~ from Homebrew...

{   'BuildPymod': False,
    'ModulePython': 'Python39Brew',
    'ModuleRuby': 'RubyVentura',
    'Platform': 'Ventura',
    'bin': 'qt5Brew.bin.macos-Ventura-release-RsysPhb39',
    'build': 'qt5Brew.build.macos-Ventura-release-RsysPhb39',
    'build_cmd': './build.sh',
    'check_cmd_only': False,
    'debug_mode': False,
    'deploy_tool': '/usr/local/opt/qt@5/bin/macdeployqt',
    'logfile': 'qt5Brew.build.macos-Ventura-release-RsysPhb39.log',
    'make_options': '--jobs=4',
    'no_qt_bindings': False,
    'no_qt_uitools': False,
    'num_parallel': 4,
    'project_dir': '/Users/kazzz-s/GitWork/klayout',
    'pyinc': '/usr/local/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/include/python3.9',
    'pylib': '/usr/local/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/lib/libpython3.9.dylib',
    'pymod_dist': {},
    'python': '/usr/local/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/bin/python3.9',
    'qmake': '/usr/local/opt/qt@5/bin/qmake',
    'rbinc': '/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/System/Library/Frameworks/Ruby.framework/Headers',
    'rbinc2': '/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/System/Library/Frameworks/Ruby.framework/Headers/ruby',
    'rblib': '/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.tbd',
    'rpath': '@executable_path/../Frameworks',
    'ruby': '/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/bin/ruby'}

##### Started deploying libraries and executables for <klayout.app> #####
 [1] Checking the status of working directory ...
     WARNING!!! You chose <-y|--deploy> while using non-OS-standard script language.
         Consider using <-Y|--DEPLOY> instead
 [2] Creating a new empty directory <HW-qt5Brew.pkg.macos-Ventura-release-RsysPhb39> for deployment ...
 [3] Creating the standard directory structure for 'klayout.app' bundle ...
 [4] Copying KLayout's dynamic link libraries to 'Frameworks' ...
 [5] Setting and changing the identification names among KLayout's libraries ...
 [6] Copying built executables and resource files ...
 [7] Setting and changing the identification names of KLayout's libraries in each executable ...
 [8] Finally, deploying Qt's Frameworks ...

 [9] Optional deployment of Python from /usr/local/opt/python@3.9/Frameworks/Python.framework ...
[... 153 lines cut ...]

[notice] A new release of pip available: 22.3.1 -> 23.2.1
[notice] To update, run: klayout -m pip install --upgrade pip
0
>>> quit()
```

13. Run the application bundle
[image]
[image]
[image]

14. Pin the python@3.9 for safe
```
(base) Ventura2{kazzz-s} ~ (1)% brew pin python@3.9

(base) Ventura2{kazzz-s} ~ (2)% brew list --pinned
python@3.8
python@3.9 <=== OK
qt@5
ruby
ruby@3.1
```

[END]

Kazzz-S, 2023-10-05:
## 5. My idea for solving this problem ##

The original implementation (thanks to Thomas) assumes that absolute paths specify library dependencies.
Therefore, the most straightforward modification would be to add Step 2 as follows.

* Step 1: Duplicate the entire Python framework into a target directory.
* Step 2: In the target directory, temporarily change the Python library dependencies from relative paths (if found) to absolute ones.
* Step 3: Do the rest of the work, including re-linking between libraries.

Because of this change, Step 3 basically does not need to be changed. That is, less error-prone.

I have added the **Change_Python_LibPath_RelativeToAbsolute()** function that works as follows.
The above pull request includes this.

```
:
:
   [9.2.1] Patching Python Framework
:
:
In [2] of Change_Python_LibPath_RelativeToAbsolute()

Edited the debug info with:
    $LOADER_PATH=@loader_path/../../../../../../../../../..

    $PYTHON_LIB=/Users/kazzz-s/GitWork/klayout/HW-qt5Brew.pkg.macos-Ventura-release-RsysPhb39/+
                klayout.app/Contents/Frameworks/Python.framework/Versions/3.9/lib/python3.9/lib-dynload

  executing: /usr/bin/install_name_tool -change  <=== 'install_name_tool' command
    $LOADER_PATH/opt/gdbm/lib/libgdbm.6.dylib    <=== original relative path (old name)
    /usr/local/opt/gdbm/lib/libgdbm.6.dylib      <=== absolute path (new name)
    $PYTHON_LIB/_dbm.cpython-39-darwin.so        <=== target library file

  executing: /usr/bin/install_name_tool -change
    $LOADER_PATH/opt/gdbm/lib/libgdbm_compat.4.dylib
    /usr/local/opt/gdbm/lib/libgdbm_compat.4.dylib
    $PYTHON_LIB/_dbm.cpython-39-darwin.so

  executing: /usr/bin/install_name_tool -change
    $LOADER_PATH/opt/mpdecimal/lib/libmpdec.3.dylib
    /usr/local/opt/mpdecimal/lib/libmpdec.3.dylib
    $PYTHON_LIB/_decimal.cpython-39-darwin.so

  executing: /usr/bin/install_name_tool -change
    $LOADER_PATH/opt/gdbm/lib/libgdbm.6.dylib
    /usr/local/opt/gdbm/lib/libgdbm.6.dylib
    $PYTHON_LIB/_gdbm.cpython-39-darwin.so

  executing: /usr/bin/install_name_tool -change
    $LOADER_PATH/opt/openssl@3/lib/libssl.3.dylib
    /usr/local/opt/openssl@3/lib/libssl.3.dylib
    $PYTHON_LIB/_hashlib.cpython-39-darwin.so

  executing: /usr/bin/install_name_tool -change
    $LOADER_PATH/opt/openssl@3/lib/libcrypto.3.dylib
    /usr/local/opt/openssl@3/lib/libcrypto.3.dylib
    $PYTHON_LIB/_hashlib.cpython-39-darwin.so

  executing: /usr/bin/install_name_tool -change
    $LOADER_PATH/opt/xz/lib/liblzma.5.dylib
    /usr/local/opt/xz/lib/liblzma.5.dylib
    $PYTHON_LIB/_lzma.cpython-39-darwin.so

  executing: /usr/bin/install_name_tool -change
    $LOADER_PATH/opt/sqlite/lib/libsqlite3.0.dylib
    /usr/local/opt/sqlite/lib/libsqlite3.0.dylib
    $PYTHON_LIB/_sqlite3.cpython-39-darwin.so

  executing: /usr/bin/install_name_tool -change
    $LOADER_PATH/opt/openssl@3/lib/libssl.3.dylib
    /usr/local/opt/openssl@3/lib/libssl.3.dylib
    $PYTHON_LIB/_ssl.cpython-39-darwin.so

  executing: /usr/bin/install_name_tool -change
    $LOADER_PATH/opt/openssl@3/lib/libcrypto.3.dylib
    /usr/local/opt/openssl@3/lib/libcrypto.3.dylib
    $PYTHON_LIB/_ssl.cpython-39-darwin.so

  executing: /usr/bin/install_name_tool -change
    $LOADER_PATH/opt/readline/lib/libreadline.8.dylib
    /usr/local/opt/readline/lib/libreadline.8.dylib
    $PYTHON_LIB/readline.cpython-39-darwin.so

           ---> Change_Python_LibPath_RelativeToAbsolute(): Changed the library paths.
:
:
```

Kazzz-S, 2023-10-05:
## 6. Re-upgrade Homebrew Python 3.9.17 ---> Python 3.9.18 ##

**Steps followed:**
```
Ventura2{kazzz-s} klayout (1)% brew list --pinned
python@3.8
python@3.9
qt@5
ruby
ruby@3.1

Ventura2{kazzz-s} klayout (2)% brew unpin python@3.8 python@3.9 qt@5

Ventura2{kazzz-s} klayout (3)% brew list --pinned
ruby
ruby@3.1

Ventura2{kazzz-s} klayout (4)% brew install python@3.9
Warning: python@3.9 3.9.18 is already installed.
The currently linked version is: 3.9.17_1

Ventura2{kazzz-s} klayout (5)% brew unlink python@3.9
Unlinking /usr/local/Cellar/python@3.9/3.9.17_1... 11 symlinks removed.

Ventura2{kazzz-s} klayout (6)% brew install python@3.9
Warning: python@3.9 3.9.18 is already installed, it's just not linked.
To link this version, run:
  brew link python@3.9

Ventura2{kazzz-s} klayout (7)% brew link python@3.9
Linking /usr/local/Cellar/python@3.9/3.9.18...
Error: Could not symlink bin/pip3
Target /usr/local/bin/pip3
already exists. You may want to remove it:
  rm '/usr/local/bin/pip3'

To force the link and overwrite all conflicting files:
  brew link --overwrite python@3.9

To list all files that would be deleted:
  brew link --overwrite --dry-run python@3.9

Ventura2{kazzz-s} klayout (8)% rm '/usr/local/bin/pip3'

Ventura2{kazzz-s} klayout (9)% brew link python@3.9
Linking /usr/local/Cellar/python@3.9/3.9.18... 18 symlinks created.

Ventura2{kazzz-s} klayout (10)% which python3
/usr/local/opt/python@3.9/bin/python3

Ventura2{kazzz-s} klayout (11)% python3
Python 3.9.18 (main, Aug 24 2023, 21:32:48) <=== OK
[Clang 14.0.3 (clang-1403.0.22.14.1)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>>

Ventura2{kazzz-s} klayout (12)% brew pin python@3.8 python@3.9 qt@5

Ventura2{kazzz-s} klayout (13)% brew list --pinned
python@3.8
python@3.9
qt@5
ruby
ruby@3.1

Ventura2{kazzz-s} macbuild (14)% cd /usr/local/opt

Ventura2{kazzz-s} opt (15)% ll | grep python
lrwxr-xr-x  1 kazzz-s admin   27  9  8 21:00 python@3.8 -> ../Cellar/python@3.8/3.8.18
lrwxr-xr-x  1 kazzz-s admin   27 10  3 21:57 python@3.9 -> ../Cellar/python@3.9/3.9.18 <=== OK

Ventura2{kazzz-s} klayout (16)% which pip3
/usr/local/opt/python@3.9/bin/pip3

Ventura2{kazzz-s} klayout (17)% pip3 --version
pip 22.3.1 from /usr/local/lib/python3.9/site-packages/pip (python 3.9)
```

Kazzz-S, 2023-10-05:
## 7. Make the HW-packages with Homebrew Python 3.9.18 ##

* Successfully made **HW-klayout-0.28.12-macOS-${OSType}-1-qt5Brew-RsysPhb39.dmg** on **BigSur**, **Monterey**, and **Ventura**. The images below were taken in Monterey.

[image]
[image]

klayoutmatthias, 2023-10-05:
Hi @Kazzz-S

Your efforts are sincerely appreciated! 

Are you implying that there are images already worth uploading the images to the server? 
Let me know if you like me to publish something.

Best regards,

Matthias

Kazzz-S, 2023-10-05:
Hi @klayoutmatthias,

> Are you implying that there are images already worth uploading the images to the server?
> Let me know if you like me to publish something.

I don't think there are any major differences between the 3.9.17 and 3.9.18 versions.
While testing, I built almost 40 DMGs with different combinations of OS and Python versions.
But I deleted them all to save disk space.
So, I have no plans to upload new images (the later ones) to the server.
This patchwork is for the next maintenance release.

Best regards,
Kazzz-S

Kazzz-S, 2023-10-06:
Closed via #1498

closed 2023-10-06 by Kazzz-S