KLayout/klayout #1489: WTF?The uninstall program broke my system

Eify666666, 2023-09-22:
The software was placed in the root directory, and after uninstalling the program to run, my C drive was almost wiped clean. What rubbish design?

Eify666666, 2023-09-22:
0.28.12 version

sebastian-goeldi, 2023-09-24:
Could you elaborate a bit more what you did exactly? You are talking about root and C:\ in one sentence, but one is a linux terminology and one a windows one, this doesn't really make sense. Furthermore did you use official installers or did you something non-standard like the tar.gz folders?

I have installed and uninstalled klayout countless times and even if I did non-standard stuff like install with `sudo make install` or `sudo make uninstall` it never just "wiped" my device.

klayoutmatthias, 2023-09-24:
@Eify666666 I'm not taking this seriously. The installer/uninstaller is NSIS which is a well-established system and the design follows the usual conventions. The installation doesn't even require admin rights, so basically I wonder how many failsafes you need to bypass to end up that way. It's a user fault, period.

klayoutmatthias, 2023-09-24:
I have placed this note on the download page (https://www.klayout.de/build.html): 

[image]

That is all I can do for now.

Eify666666, 2023-09-27:
The following is how the accident happened:
1.I install the klayout in 'C:\' in a windows system PC.
2.I notic that it don't create subfolders during installation, such as' c:\klayout\ '.So I need to uninstall, and re-install.
3.Uninstall is by the control panel, shown in the picture. And then, the program deletes all files in disk C in alphabetical order.

[image]

chkkbim, 2023-09-29:
[1] I found a warning in the NSIS wiki that it may not be safe to use RMDir /r $INSTDIR :

chkkbim, 2023-09-29:
[1] I found a warning (dated Nov 2017) in the NSIS wiki that using `RMDir /r $INSTDIR` may not be safe :
[https://nsis.sourceforge.io/Reference/RMDir](url)

[2] Their suggested solution is at :
[https://nsis.sourceforge.io/Uninstall_only_installed_files](url)

[3] Some users of other project have run into similar problems : 
[https://github.com/electron-userland/electron-builder/issues/3201](url)

[4] Perhaps a way to avoid these unhappy issues is to manually remove the "uninstall.exe" from the package before offering for download ?

HTH

whisperity, 2023-10-16:
@chkkbim To be fair, it does look like uninstalling something in a way that it works in the most edge case of environment ever is a yet unsolved mystery of computer science, sometimes not even specific to Windows things...

 * http://github.com/MrMEEE/bumblebee-Old-and-abbandoned/issues/123
 * http://github.com/ValveSoftware/steam-for-linux/issues/3671