KLayout/klayout #1490: TAPERRULE incorrrectly implemented in DEF reads

RTimothyEdwards, 2023-09-26:
In the attached DEF file:
[openframe_project_wrapper.def.gz](https://github.com/KLayout/klayout/files/12729338/openframe_project_wrapper.def.gz)
Nets have been assigned to each pin and each net has been placed directly under each pin.  Each net is drawn using a NONDEFAULT rule so that the net's route is exactly the width of the pin.

The issue is that when this DEF file is read into klayout, the nets are wire-extended past the ends of the pins (see, for example, the pin "mask_rev[31]").  That indicates that klayout has assumed a 1/2 wire width extension for the wire.  However, the DEF spec states that NONDEFAULT rules have a default wire extension of zero when WIREEXT is not specified for the NONDEFAULT rule layer.  Since the attached DEF file does not specify any WIREEXT number for any layer in the NONDEFAULT rules, then no wire extension should be added to these wires.

stefanottili, 2023-09-26:
Which part of the def spec are are you referring to ?

lefdefref58.pdf page 152

WIREEXTENSION value

Specifies the distance by which wires are extended at vias. Enter 0 (zero) to specify no extension. Values other than 0 must be greater than or equal to half of the routing width for the layer, as defined in the nondefault rule.
**Default: Wires are extended half of the routing width**

openroad agree's with klayout's implementation

openroad -gui taper.or

cat taper.or
read_lef ~/OpenROAD-flow-scripts/flow/platforms/sky130hd/lef/sky130_fd_sc_hd.tlef
read_def -continue_on_errors openframe_project_wrapper.def.gz

Is it correct to assume that you're the def is written by magic and that the technology is defined in
OpenROAD-flow-scripts/flow/platforms/sky130hd/lef/sky130_fd_sc_hd.tlef ?

The def is unusual in a couple of area's:

a) pins are outside the diearea
b) usually detailed routing to pins end in the middle of the pin
c) it's unusual for NETS not to be mostly nondefault rule, even when connecting to larger pins
d) metal width is not defined for all layers for each rule ( OR read_def needs -continue_on_errors)

the def only shows the net portion overlapping the pin, it would be good to know how they continue

RTimothyEdwards, 2023-09-26:
@stefanottili :  I have the 5.7 spec in front of me, not the 5.8 spec, but it's not the WIREEXTENSION definition, it's the definition of "NONDEFAULTRULES" rule "+ LAYER WIREEXT".  (Page 270 of the 5.7 spec):

```
WIREEXT _wireExt_     Specifies the distance by which wires are extended at vias on _layerName_.
                                     _Default:_ 0
                                     _Type:_ Integer, specified in DEF database units.
```
So the NONDEFAULT rule differs from regular wiring rules by having a default wire extension of 0 instead of 1/2 wire width.

stefanottili, 2023-09-26:
5.8 page 278, Nondefault Rules 
```
WIREEXT wireExt
Specifies the distance by which wires are extended at vias. Enter 0 (zero) to specify no extension. Values other than 0 must be greater than or equal to half of the routing width for the layer, as defined in the nondefault rule.
Default: Wires are extended half of the routing width
```

RTimothyEdwards, 2023-09-26:
@stefanottili :  I can't believe that the spec changed the default between versions without backwards compatibility. . .   So in 5.7, the default is zero and in 5.8, the default is 1/2 wire width.  Ugh.

RTimothyEdwards, 2023-09-26:
@stefanottili :  I should point out, though, that the DEF file that I attached does, in fact, specify version 5.7 at the top, so klayout is also incorrectly handling the default value for the DEF spec version.

stefanottili, 2023-09-26:
a) I don’t think your interpretation of the 5.7 spec is correct.

b) > wireExt: Specifies the distance by which wires are extended at vias

nets always have half width extension at begin and end.
If you want 0 length begin/end, put the net geometry in the specialnet section.

RTimothyEdwards, 2023-09-26:
@stefanottili :  Yes, you are right and I failed to note that the extension is only at vias.  Thank you for pointing that out.
I do have a lot of unusual routing, mostly to cover a bunch of use cases.  This is a routine converting arbitrary layout to DEF, where the layout may not have anything to do with a place & route tool but the DEF file needs to accurately represent the layout anyway.  I am generally satisfied if I can produce a DEF file that matches the layout, regardless of how weird it is.

RTimothyEdwards, 2023-09-26:
@stefanottili :  If you wish to pursue any of the issues you mentioned related to WIREEXT properly used as a via extension only, then it doesn't really match the intent of this issue as posted.  Since this issue has been determined not to be a problem in klayout, I will go ahead and close the issue.

closed 2023-09-26 by RTimothyEdwards