KLayout/klayout #1492: Mac DMGs for 0.28.12

Kazzz-S, 2023-09-26:
Dear @klayoutmatthias,

Thank you for the maintenance release.  Sorry about my late response.
As usual, different DMG packages of **KLayout 0.28.12 for Mac** are available as described below ([0.28.12 Dropbox Link](https://www.dropbox.com/scl/fo/7q9i3tq9ywh5rwrmu9qk7/h?rlkey=oqesqtzekp7e3qiyp4jm7jeoi&dl=0)).
However, **four DMGs built on **Ventura** are NOT available this time.**
Instead, I expect DMGs built on Monterey will be usable in Ventura. I will comment on that [here](https://www.klayout.de/forum/discussion/2377/).

## DMG file naming convention ##
[image]

where
(a) Type of package:  ST-: standard; LW-: lightweight; HW-: heavyweight
(b) KLayout version
(c) Target OS
(d) Package serial number
(e) Qt5 type:  qt5MP: MacPorts; qt5Brew: Homebrew; qt5Ana3: Anaconda3
(f) **R**uby type:   sys: OS-bundled; mp32: Ruby3.2 from MacPorts; hb32: Ruby3.2 from Homebrew; ana3: Ruby3.1 from Anaconda3
(f) **P**ython type: sys: OS-bundled; mp39: Python 3.9 from MacPorts; hb39: Python 3.9 from Homebrew; ana3: Python3.9 from Anaconda3

## For the primary platform, macOS Catalina (10.15.7) ##
Five macOS DMGs for **Catalina** are available.
They were built on an Intel-CPU-based (not M1 chip) VMware Fusion 13.0.2 on Monterey (12.7).

**LW- and HW- packages use Python 3.9 from MacPorts, Homebrew, or Anaconda3.**
|Srl.No.|Target OS|DMG file name|Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
1|Catalina (10.15)|ST-klayout-0.28.12-macOS-Catalina-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(1)|
2|Catalina (10.15) with MacPorts dev. env.|LW-klayout-0.28.12-macOS-Catalina-1-qt5MP-Rmp32Pmp39.dmg|MacPorts|MacPorts|MacPorts|(2)|
3|Catalina (10.15) with Homebrew dev. env.|LW-klayout-0.28.12-macOS-Catalina-1-qt5Brew-Rhb32Phb39.dmg|Homebrew|Homebrew|Homebrew|(3)|
4|Catalina (10.15) with Anaconda3 dev. env.|LW-klayout-0.28.12-macOS-Catalina-1-qt5Ana3-Rana3Pana3.dmg|Anaconda3|Anaconda3|Anaconda3|(4)|
5|Catalina (10.15)|HW-klayout-0.28.12-macOS-Catalina-1-qt5Brew-RsysPhb39.dmg|Homebrew|OS-bundled|Homebrew|(5)|

(1) shares OS-bundled Ruby2.6 and Python-2.7; Qt5 from MacPorts is embedded, which is the standard (ST-).
(2) **experimental**; shares MacPorts development environment; for those with the matching dev. env.
(3) **experimental**; shares Homebrew development environment; for those who have the matching dev. env.
(4) **experimental**; shares Anaconda3 development environment; for those who have the matching dev. env.
(5) **experimental**; shares OS-bundled Ruby2.6; Qt5 and Python-3.9 from Homebrew are embedded, hence, heavy.

## For macOS Big Sur (11.7.10) ##
Four macOS DMGs for **Big Sur** are available.
They were built on an Intel-CPU-based (not M1 chip) VMware Fusion 13.0.2 on Monterey (12.7).

**LW- and HW- packages use Python 3.9 from MacPorts, Homebrew, or Anaconda3.**
|Srl.No.|Target OS|DMG file name|Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
21a|~~BigSur (11.7)~~|~~ST-klayout-0.28.12-macOS-BigSur-1-qt5MP-RsysPsys.dmg~~|~~MacPorts~~|~~OS-bundled~~|~~OS-bundled~~|(21a)|
21b|Catalina (10.15)|ST-klayout-0.28.12-macOS-Catalina-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(21b)|
22|BigSur (11.7) with MacPorts dev. env.|LW-klayout-0.28.12-macOS-BigSur-1-qt5MP-Rmp32Pmp39.dmg|MacPorts|MacPorts|MacPorts|(22)|
23|BigSur (11.7) with Homebrew dev. env.|LW-klayout-0.28.12-macOS-BigSur-1-qt5Brew-Rhb32Phb39.dmg|Homebrew|Homebrew|Homebrew|(23)|
24|BigSur (11.7) with Anaconda3 dev. env.|LW-klayout-0.28.12-macOS-BigSur-1-qt5Ana3-Rana3Pana3.dmg|Anaconda3|Anaconda3|Anaconda3|(24)|
25|BigSur (11.7)|HW-klayout-0.28.12-macOS-BigSur-1-qt5Brew-RsysPhb39.dmg|Homebrew|OS-bundled|Homebrew|(25)|

(21a) **NOT PROVIDED**; The ST-* package is no longer provided since Xcode >= 13.2 does not link the legacy Python 2.7 library.
(21b) However, ST*.dmg for Catalina, which shares OS-bundled Ruby2.6 and Python-2.7, and Qt5 from MacPorts is embedded, can run on Big Sur.
(22) **experimental**; shares MacPorts development environment; for those who have the matching dev. env.
(23) **experimental**; shares Homebrew development environment; for those who have the matching dev. env.
(24) **experimental**; shares Anaconda3 development environment; for those who have the matching dev. env.
(25) **experimental**; shares OS-bundled Ruby2.6; Qt5 and Python-3.9 from Homebrew are embedded, hence, heavy.

## For macOS Monterey (12.7) ##
Four macOS DMGs for **Monterey** are available.

**LW- and HW- packages use Python 3.9 from MacPorts, Homebrew, or Anaconda3.**
|Srl.No.|Target OS|DMG file name|Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
31a|~~Monterey (12.7)~~|~~ST-klayout-0.28.12-macOS-Monterey-1-qt5MP-RsysPsys.dmg~~|~~MacPorts~~|~~OS-bundled~~|~~OS-bundled~~|(31a)|
31b|~~Catalina (10.15)~~|~~ST-klayout-0.28.12-macOS-Catalina-1-qt5MP-RsysPsys.dmg~~|~~MacPorts~~|~~OS-bundled~~|~~OS-bundled~~|(31b)|
32|Monterey (12.7) with MacPorts dev. env.|LW-klayout-0.28.12-macOS-Monterey-1-qt5MP-Rmp32Pmp39.dmg|MacPorts|MacPorts|MacPorts|(32)|
33|Monterey (12.7) with Homebrew dev. env.|LW-klayout-0.28.12-macOS-Monterey-1-qt5Brew-Rhb32Phb39.dmg|Homebrew|Homebrew|Homebrew|(33)|
34|Monterey (12.7) with Anaconda3 dev. env.|LW-klayout-0.28.12-macOS-Monterey-1-qt5Ana3-Rana3Pana3.dmg|Anaconda3|Anaconda3|Anaconda3|(34)|
35|Monterey (12.7)|HW-klayout-0.28.12-macOS-Monterey-1-qt5Brew-RsysPhb39.dmg|Homebrew|OS-bundled|Homebrew|(35)|

(31a) **NOT PROVIDED**; The ST-* package is no longer provided since Xcode >= 13.2 does not link the legacy Python 2.7 library.
(31b) **Monterey 12.3.1 finally eliminated Python-2.7 binary. Hence, ST-\* package no longer runs.**
(32) **experimental**; shares MacPorts development environment; for those who have the matching dev. env.
(33) **experimental**; shares Homebrew development environment; for those who have the matching dev. env.
(34) **experimental**; shares Anaconda3 development environment; for those who have the matching dev. env.
(35) **experimental**; shares OS-bundled Ruby2.6; Qt5 and Python-3.9 from Homebrew are embedded, hence, heavy.

## For the latest macOS Ventura (13.6) ##
Four macOS DMGs for **Ventura** are NOT available this time.
Due to the issue [captured here](https://www.klayout.de/forum/discussion/2377/), they were NOT successfully built with the latest Xcode 15.0.
The development environment is in an Intel-CPU-based (not M1 chip) VMware Fusion 13.0.2 on Monterey (12.7).

**LW- and HW- packages use Python 3.9 from MacPorts, Homebrew, or Anaconda3.**
|Srl.No.|Target OS|DMG file name|Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
41|~~Ventura (13.6) with MacPorts dev. env.~~|~~LW-klayout-0.28.12-macOS-Ventura-1-qt5MP-Rmp32Pmp39.dmg~~|~~MacPorts~~|~~MacPorts~~|~~MacPorts~~|~~(41)~~|
42|~~Ventura (13.6) with Homebrew dev. env.~~|~~LW-klayout-0.28.12-macOS-Ventura-1-qt5Brew-Rhb32Phb39.dmg~~|~~Homebrew~~|~~Homebrew~~|~~Homebrew~~|~~(42)~~|
43|~~Ventura (13.6) with Anaconda3 dev. env.~~|~~LW-klayout-0.28.12-macOS-Ventura-1-qt5Ana3-Rana3Pana3.dmg~~|~~Anaconda3~~|~~Anaconda3~~|~~Anaconda3~~|~~(43)~~|
44|~~Ventura (13.6)~~|~~HW-klayout-0.28.12-macOS-Ventura-1-qt5Brew-RsysPhb39.dmg~~|~~Homebrew~~|~~OS-bundled~~|~~Homebrew~~|~~(44)~~|

(41) **experimental**; shares MacPorts development environment; for those who have the matching dev. env.
(42) **experimental**; shares Homebrew development environment; for those who have the matching dev. env.
(43) **experimental**; shares Anaconda3 development environment; for those who have the matching dev. env.
(44) **experimental**; shares OS-bundled Ruby2.6; Qt5 and Python-3.9 from Homebrew are embedded, hence, heavy.

## For previous OS ##
|Srl.No.|Target OS|DMG file name|Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
6|Mojave (10.14)|From 0.27.x onward, DMGs will not be provided for this OS.||||(1p)|
7|HighSierra (10.13)|From 0.27.x onward, DMGs will not be provided for this OS.||||(1p)|
8|Sierra (10.12)|From 0.27.x onward, DMGs will not be provided for this OS.||||(1p)|
9|ElCapitan (10.11)|From 0.27.x onward, DMGs will not be provided for this OS.||||(1p)|
x|Yosemite (10.10)|has been dropped due to the compiler issue reported in #360.|

(1p) I expect that build is still possible.

Please close this ticket after publishing the DMGs.
With warm regards,
Kazzz-S

klayoutmatthias, 2023-09-27:
Perfect! Files are on the server :)

I understand that was an exceptional effort in this case. I hope the problems are solved and it will not be that difficult on the next release.

Many thanks for your continued effort!

Best regards,

Matthias

closed 2023-09-27 by klayoutmatthias

Kazzz-S, 2023-09-28:
Dear @klayoutmatthias,

I could also build the DMGs on Ventura by applying the patch ( #1493 ).
Four DMG packages of **KLayout 0.28.12 for Mac** built on Ventura are available as described below ([0.28.12 Dropbox Link](https://www.dropbox.com/scl/fo/7q9i3tq9ywh5rwrmu9qk7/h?rlkey=oqesqtzekp7e3qiyp4jm7jeoi&dl=0)).
Please append those DMGs to the server.

## For the latest macOS Ventura (13.6) ##
Four macOS DMGs for **Ventura** are available.
They were built on an Intel-CPU-based (not M1 chip) VMware Fusion 13.0.2 on Monterey (12.7).

**LW- and HW- packages use Python 3.9 from MacPorts, Homebrew, or Anaconda3.**
|Srl.No.|Target OS|DMG file name|Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
41|Ventura (13.6) with MacPorts dev. env.|LW-klayout-0.28.12-macOS-Ventura-1-qt5MP-Rmp32Pmp39.dmg|MacPorts|MacPorts|MacPorts|(41)|
42|Ventura (13.6) with Homebrew dev. env.|LW-klayout-0.28.12-macOS-Ventura-1-qt5Brew-Rhb32Phb39.dmg|Homebrew|Homebrew|Homebrew|(42)|
43|Ventura (13.6) with Anaconda3 dev. env.|LW-klayout-0.28.12-macOS-Ventura-1-qt5Ana3-Rana3Pana3.dmg|Anaconda3|Anaconda3|Anaconda3|(43)|
44|Ventura (13.6)|HW-klayout-0.28.12-macOS-Ventura-1-qt5Brew-RsysPhb39.dmg|Homebrew|OS-bundled|Homebrew|(44)|

(41) **experimental**; shares MacPorts development environment; for those who have the matching dev. env.
(42) **experimental**; shares Homebrew development environment; for those who have the matching dev. env.
(43) **experimental**; shares Anaconda3 development environment; for those who have the matching dev. env.
(44) **experimental**; shares OS-bundled Ruby2.6; Qt5 and Python-3.9 from Homebrew are embedded, hence, heavy.

Kazzz-S

klayoutmatthias, 2023-09-30:
Thanks for pointing me to the new downloads - as the ticket was already closed, I did not receive an update. I have uploaded the new files to the server.

Again, many thanks for your efforts!

Best regards,

Matthias