KLayout/klayout #1494: must-have Python modules for Windows

klayoutmatthias, 2023-09-30:
This is a placeholder for collecting a list of must-have Python modules I should include in the Windows distribution. Please comment and tell what modules you think are must-have ones. If there a multiple modules doing similar things, I'd like to include only the most popular one.

This is not a commitment to include all of them, but maybe it is possible to finally vote on a list of additional modules.

Kazzz-S, 2023-10-02:
Thanks for your survey.
I want to vote for the modules below to manipulate plane text CSV files and Excel books.
* pandas
* XlsxWriter

ernie14674560, 2023-10-02:
Hi Matthias: 
Thanks for your effort.
I want to include the following Python module to make my work easier.
-->  python-pptx

klayoutmatthias, 2024-01-05:
I have added xlsxwriter and pptx to the 0.28.15 Windows binary distributions. I'm closing this ticket now. Please open a new ticket if you think a certain module was good to have too.

closed 2024-01-05 by klayoutmatthias

Kazzz-S, 2024-09-02:
Hello @klayoutmatthias,

I'd like to add
* openpyxl

I found this module is required when I wrote a sample script at https://www.klayout.de/forum/discussion/2579/.

klayoutmatthias, 2024-09-02:
It's noted :)

I will try to add that module.

Matthias

klayoutmatthias, 2024-09-23:
Hi @Kazzz-S 

openpyxl is now included in the latest (0.29.7) Windows release.

Best regards,

Matthias

Kazzz-S, 2024-09-23:
Hi @klayoutmatthias,

Many thanks!

Best regards,
Kazzz-S