KLayout/klayout #1500: MacOS Support (Sanoma 14.0)

yakshsheoran, 2023-10-05:
Hello to the volunteers who provide an invaluable MacOS support for Klayout. I'm trying to the software for the first time, as I need it at my new job, however there is no version on the website for the MacOS Sanoma 14.0. I implore the few users who delicate their own hardware and spare time for MacOS support, to help me with the issue. 

I would really appreciate any help, since this is a new software for me. 

Kind regards,
Yaksh

Kazzz-S, 2023-10-05:
Hi Yaksh,

I've created these DMGs for Intel (x86_64) Macs. 
I have not yet installed the latest Sonoma (14.0) on my Intel Mac and have no plans to do so for the time being.

If you have an Intel Mac, try one built on Ventura.
Specifically, **HW-klayout-0.28.12-macOS-Ventura-1-qt5Brew-RsysPhb39.dmg**, which does not require a development environment like Homebrew.
In contrast, DMGs named **LW-** do require a corresponding development environment.

If you use an Apple Silicon (M1 or M2) Mac, you must install **Rosetta 2** to run an executable built for x86_64.
However, I have no experience with this as I do not own an Apple silicon machine :-(

Warm regards,
Kazzz-S

yakshsheoran, 2023-10-05:
Hello Kazzz, 
Thank you for your contributions. I use an Apple Sillicon Mac (M1). I checked my OS again turns out I still have the Big Sur 11.7.4. Still not entirely sure which file I should use? Should i also download Rosetta 2 to run files for Intel Macs?

Kind regards,
Yaksh

Kazzz-S, 2023-10-05:
Hi Yaksh,

Then, please attempt **HW-klayout-0.28.12-macOS-BigSur-1-qt5Brew-RsysPhb39.dmg** with Rosetta 2.
I repeat that I have yet to experience with Rosetta 2. I have only heard rumors that they use Rosetta 2 to use these DMGs.

Best regards,
Kazzz-S

yakshsheoran, 2023-10-05:
I installed Rosetta 2 and the installed file still appears to be damaged. I dont think it is a file that works with rosetta 2. However the app that runs after installation is called KLayoutHomebrew-H. Is this the correct name? 

Kind regards,
Yaksh

Kazzz-S, 2023-10-05:
1. Could you check if the MD5 value matches below? Otherwise, damaged.
```
40eb38111e581271a9bd19c6c9c319e5 *HW-klayout-0.28.12-macOS-BigSur-1-qt5Brew-RsysPhb39.dmg
```

2. Have you dragged & dropped the **klayout.app** icon shown below? 
    Then, the application name should be "klayout.app"
[image]

Kazzz-S

yakshsheoran, 2023-10-05:
Oh wonderful, I figured it out, it opens now. However the system cannot verify if the app is free from malware and therefore, wont work. Any advice on fixing this? 

Yaksh

Kazzz-S, 2023-10-05:
For example, https://iboysoft.com/howto/cannot-be-opened-because-it-is-from-an-unidentified-developer.html

Kazzz-S

yakshsheoran, 2023-10-05:
Thanks for your help. It all worked out. 

Yaksh

klayoutmatthias, 2023-10-05:
Wonderful :) I'll leave that ticket open for a while so others may benefit from these insights too.

Matthias