KLayout/klayout #1505: RBA::Shapes.insert() questions

dsus4wang, 2023-10-07:
Hi Matthias,
Recently I encountered some issues about Shapes.insert()
I have checked the document that Shapes.insert(Region region) has no return (void), But actually when I run a Shapes.insert(region), It returned a Shapes object[1] pointing to the shapes container which the region insert to. Then I inserted another Region into the shapes container, It returned a shapes object[2].
Strangely, [1] and [2] have the different hash but they are pointing to the same shapes container, I think maybe it is a bug?
Looking for your reply,
Wang

klayoutmatthias, 2023-10-08:
That is a kind of convenience hack. When a function is declared "void" in C++, in Ruby/Python it will return a reference to self (or something like that, see below), to allow chaining calls.

That is not consistent in the doc, I have not fixed that yet.

Regarding the returned object: all Python/Ruby objects you see in RBA/pya are actually proxies to C++ objects. You can consider them pointers. For some objects, KLayout will stash the Python/Ruby object and reuse it - so whenever you reference to a specific C++ object, it will return the same Python or Ruby object. In order to do so, it need to keep the information along with the C++ object and that add a number of bytes and comes with a runtime penalty. So that is something that is only done for heavy objects such as Layout. Shapes is a lightweight container and for these, all references are turned into different proxy objects. So you do not need to worry about these objects being different - they still point to the same C++ object. 

That is not a bug but some implementation detail. Shapes does not provide a specific hash value, so I assume it is taken from the object ID and that is a different one for each individual proxy.

Matthias

dsus4wang, 2023-10-09:
Thanks for your reply Matthias, here's another question: In some cases I need to check if the two shapes containers point to the same c++ object. Now I can only use Shapes.each and then compare each shapes in that container, but its time cost is so large. So I wonder if there's other way to check if the two shapes containers point to the same c++ object?
Wang

klayoutmatthias, 2023-10-11:
Hi Wang,

no, there is no specific method to check this. The C++ identity is an opaque property. You have to redesign your code.

Basically, the tuple ```(cell_index, layer_id)``` is representative for the Shapes container inside the same layout. Both members of the tuple are integers and uniquely identify the cell and layer. This unique identification is actually the reason why there is a cell index. So if you store that tuple instead of the Shapes reference, you can easily get the Shapes container from it:

```
layout.cell(cell_index).shapes(layer_id) -> Shapes
```

Identity of the tuple will tell you that you refer to the same container.

Matthias

dsus4wang, 2023-10-14:
Thanks again for your help!

closed 2023-10-14 by dsus4wang