KLayout/klayout #1509: Wrong result of boolean or in EdgeProcessor branch rather than clip_poly branch

Jebearssica, 2023-10-11:
Hi, I may find a problem in EdgeProcessor. When I calculate layer1 & layer2, the result is not correct as follows:
[image]

However, when I delete a shape in layer1, (i.e. layer1 becomes a box), the result is correct since the workflow is clip_poly not EdgeProcessor. Such behavior may point out that somewhere in EdgeProcessor is wrong?

I try to figure out but failed, and I have done the following test:

* when DBU is 1e-4 or 1e-1, same behavior will reproduced. (the ouput will not clip the outside 1 DBU triangle)
* when DBU is 1e-3, can not reproduce such behavior.
* try to replace double cast to db::Coord with `round` in EdgeProcessor, nothing happen.

Here's the demo oas file, layer2 contains 2 small boxes. The bottom box is completely inside layer1, while the other one has a corner outside layer1.
[output_error_small_demo.zip](https://github.com/KLayout/klayout/files/12865644/output_error_small_demo.zip)

klayoutmatthias, 2023-10-11:
Thanks for the test case. I can reproduce the issue with "layer_1" and "layer_space". I guess this is what you mean.

I will try to debug the problem. However, I consider that a pretty marginal issue.

klayoutmatthias, 2023-10-12:
I debugged the issue and it's not easy to fix.

The reason is that the edge processor establishes a certain fuzziness to account for snapping at edge intersections. The single-DBU outlier falls into that fuzziness interval and is ignored therefore.

Fixing the problem in general compromises the ability for a stable handling of the all-angle and manifold-intersection cases.

Jebearssica, 2023-10-19:
Thanks, I think it is pretty strange, when I set DBU to 9.5 * 1e-5, the final result is correct. But under such condition, the outlier should also be single-DBU, isn't it?

klayoutmatthias, 2023-10-19:
I meant to reduce the DBU to half so that all dimensions become even numbers in terms of DBU and the corner becomes 2x2 DBU instead of 1x1 DBU. I do not mean to change the DBU alone, but also scale the polygons in integers, so that physically you get the same layout, but internally, the corners outside of the diagonal will be either zero or 2x2 DBU.

Technically the problem is that: to achieve robustness against snapping in the all-angle case, KLayout considers all non-horizontal/non-vertical edges "fuzzy". That means it implies some halo and allows the edge to break and move inside that corridor. It also consolidates corners inside that corridor to remove edge intersections on that scale. In the 1x1 DBU case this leads to that consolidation:

[image]

the gray zone is the corridor that was mentioned. It represents the worst-case snapping hull over every point of the edge. The lower-left corner of the blue rectangle now implies a consolidation of the red edge into the two-piece dashed edge. That will remove the intersection and the boolean result of the triangle AND the rectangle is the rectangle itself. 

When the corner is 2x2 DBU, the lower-left corner of the rectangle is outside the consolidation corridor and the red edge will remain undisturbed. The intersections are computed as expected:

[image]

Matthias

closed 2023-11-20 by Jebearssica