KLayout/klayout #1513: Add Qt5 binding for QMainWindow::resizeDocks

EugeneZelenko, 2023-10-19:
Hi, Matthias!

I'll be great to provide Qt5 binding for `QMainWindow::resizeDocks`. According to [documentation](https://doc.qt.io/qt-5/qmainwindow.html#resizeDocksl), it's available since Qt 5.6. Method signature is same as in Qt6, so implementation could be shared.

klayoutmatthias, 2023-10-19:
I will try to update to Qt. I have 5.12.12 in stock which is equivalent to 5.12 LTS and no too fresh. I guess that will do it. But doing the migration is always a pain as it creates so many deltas.

klayoutmatthias, 2023-10-19:
I need to postpone this fix to 0.29 (let's just say that will happen soon ...).
Reason is that I made this transition there already and can't backport because there are too many other changes. Doing that now in 0.28 will create too many conflicts to handle.

Can this wait for, let's say two months?

Matthias

EugeneZelenko, 2023-10-19:
Hi, Matthias!

Sure, this could wait.

klayoutmatthias, 2024-03-29:
0.29 will be released soon :)

closed 2024-03-29 by klayoutmatthias