KLayout/klayout #1514: Red alert: KLayout Package Manager will stop working beginning of 2024.

klayoutmatthias, 2023-10-21:
GitHub will discontinue Subversion support, which is the basis of all the packages hosted in KLayout (https://github.blog/2023-01-20-sunsetting-subversion-support/). KLayout uses the SVN interface to download the package data.

There is need for a separate strategy, which unfortunately is not readily available. Integrating a git client may be an option. Whether it is feasible I can't say.

In any case, Package download will only work with new KLayout versions. I'm targeting 0.28.13 for support in that respect.

Matthias

lukasc-ubc, 2023-10-27:
@klayoutmatthias I tried to update a package today, and found that perhaps you have already turned off GitHub?

Would you be able to update the package manually?  siepic/SiEPIC-Tools.git/tags/v0.4.3/klayout_dot_config

[image]

Clearly I will have to migrate by January, but it would be great if I could continue using the existing GitHub release functionality until a solution is found.  Could you leave it operational at least for another month?

Thank you very much,
Lukas

klayoutmatthias, 2023-10-28:
Hi Lukas,

I have re-enabled GitHub with a deprecation warning and will do another package rescue action in a month or so. But latest Jan. 2024 packages can no longer be downloaded from GitHub the current way.

I am working on Git integration, but that is more tedious than I though initially.

Matthias

lukasc-ubc, 2023-10-29:
Thank you. 
Would it be easier to automate the process of mirroring a GitHub repo onto your server? Maybe GitHub actions could help. Or automated mirror to the other server.

klayoutmatthias, 2023-11-01:
Some update on the progress: I have enabled the Git protocol in the package manager now. It needs are special URL like "git+https://github.com/..." with some decoration to indicate branch or tag and a sub-folder for the package data (yes, that is supported). I can basically release that with the next minor release. However, it will not be useful yet as the package indexer also needs to be modified.

For this, it is required to supply full package information through the package index as pulling the package definition from Git is extremely expensive in some cases. Also the package indexer needs a cache to prevent heavy network traffic and the user interface has to support the new URL scheme. That will take some more time.

Matthias

sebastian-goeldi, 2023-11-15:
Since I saw it as a concern popping up in some context about having to clone the full working tree:

It is possible to not do that by shallow cloning. Shallow cloning will only go to a certain depth and not download the full hostory.

If I saw right, klayout is now using libgit2. libgit2 supports shallow cloning since 1.7.0 (released 4months ago). Unfortunately this means older LTS versions most likely won't have it available natively. An option would be to allow this if available in order to save bandwidth and time.

So,in the end, shallow fetch could be achieved through this:

```bash
git init
git remote add origin <url>
git fetch --depth 1 origin <sha1>
git checkout FETCH_HEAD 
```
(from [stackoverflow](https://stackoverflow.com/a/43136160))

Another more github centric approach would have been to use archive downloads for example to grab the latest commit:

http://github.com/klayout/klayout/archive/9d8e0f480a710c24be1d2f8d085845d4b807f22a.zip

Or latest (release) tag:

http://github.com/klayout/klayout/archive/v0.28.12.zip

klayoutmatthias, 2023-11-15:
@sebastian-goeldi 

Thanks for the comments. 

I considered shallow cloning already, but first of all, the libgit2 library versions out there are much older - e.g. Ubuntu 22 ships with 1.1.0 and CentOS7 comes with 0.26.8. But even if libgit2 would provider general support for shallow clones, the effect is not as large as one would expect: I gave a number of 277 MB download for a single file on one of Lukas' repos somewhere - that was already utilizing shallow clones. Without it was like 800 MB. Shallow clones still have to download the pack files which hold more than a single commit. I think this is why the improvement is behind expectation.

I'm not longer trusting the GitHub download links as they may be deprecated tomorrow and tags can addressed only if released. I think that GitHub wants us to use their API, but there is a rate limit and they are going to change it deprecate features.

There used to be a dumb protocol to talk to Git servers which features a real object ID database. That was a cool option as it allowed to retrieve single files, but I think it is gone and GitHub no longer supports it.

With the caching scheme I mentioned there is at least only one place where the download needs to happen to extract the package metadata and I guess that is acceptable. Most packages are small anyway.

Thanks and best regards,

Matthias

klayoutmatthias, 2023-12-02:
I think we can close this topic here. I hope with 0.28.13 this problem can be solved by using Git repos.