KLayout/klayout #1515: Klayout App crashed when use cell.flatten

dsus4wang, 2023-10-23:
Hi Matthias,
Now I have a GDS layout including some shapes pointing to one object(for example: 
[image]
this two polygons, if I delete one, the other one will disappear at the same time) The two polygons are in two PCells and the two PCells are in one TOP cell, Here is the problem: when I select this two polygons and then run a macro `CellView.active.layout.top_cell.flatten(true)` , Then app crashed.
Here is the crash log:
```
Signal number: 11
Address: 0xf830dc9bdc88
Program Version: KLayout 0.28.12 (2023-09-25 rLatestSourcePackage)

Backtrace:
/Applications/klayout.app/Contents/Frameworks/libklayout_lay.0.dylib +0x1fc648 _ZN3lay25enable_signal_handler_guiEb
/usr/lib/system/libsystem_platform.dylib +0x3a24 _sigtramp
/Applications/klayout.app/Contents/Frameworks/libklayout_laybasic.0.dylib +0x18384c _ZNK3lay14ObjectInstPath8is_validEPNS_14LayoutViewBaseE
/Applications/klayout.app/Contents/Frameworks/libklayout_edt.0.dylib +0xc64e4 _ZN3edt7Service20do_selection_to_viewEv
/Applications/klayout.app/Contents/Frameworks/libklayout_tl.0.dylib +0x6e9a0 _ZN2tl23DeferredMethodScheduler10do_executeEv
/Applications/klayout.app/Contents/Frameworks/libklayout_tl.0.dylib +0x7c8f8 _ZN2tl25DeferredMethodSchedulerQt5timerEv
/Applications/klayout.app/Contents/Frameworks/libklayout_tl.0.dylib +0x7c8a4 _ZN2tl25DeferredMethodSchedulerQt5eventEP6QEvent
/opt/homebrew/Cellar/qt@5/5.15.10/lib/QtWidgets.framework/Versions/5/QtWidgets +0x115d4 _ZN19QApplicationPrivate13notify_helperEP7QObjectP6QEvent
/opt/homebrew/Cellar/qt@5/5.15.10/lib/QtWidgets.framework/Versions/5/QtWidgets +0x12aa0 _ZN12QApplication6notifyEP7QObjectP6QEvent
/Applications/klayout.app/Contents/Frameworks/libklayout_lay.0.dylib +0x7b2e8 _ZN3lay14GuiApplication9do_notifyEP7QObjectP6QEvent
/opt/homebrew/Cellar/qt@5/5.15.10/lib/QtCore.framework/Versions/5/QtCore +0x1ce410 _ZN16QCoreApplication15notifyInternal2EP7QObjectP6QEvent
/opt/homebrew/Cellar/qt@5/5.15.10/lib/QtCore.framework/Versions/5/QtCore +0x1cf524 _ZN23QCoreApplicationPrivate16sendPostedEventsEP7QObjectiP11QThreadData
/opt/homebrew/Cellar/qt@5/5.15.10/plugins/platforms/libqcocoa.dylib +0x3dbd8 qt_plugin_instance
/opt/homebrew/Cellar/qt@5/5.15.10/plugins/platforms/libqcocoa.dylib +0x3e490 qt_plugin_instance
/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation +0x7f5cc __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__
/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation +0x7f560 __CFRunLoopDoSource0
/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation +0x7f2d0 __CFRunLoopDoSources0
/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation +0x7ded8 __CFRunLoopRun
/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation +0x7d448 CFRunLoopRunSpecific
/System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/HIToolbox.framework/Versions/A/HIToolbox +0x30df0 RunCurrentEventLoopInMode
/System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/HIToolbox.framework/Versions/A/HIToolbox +0x30a80 ReceiveNextEventCommon
/System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/HIToolbox.framework/Versions/A/HIToolbox +0x30984 _BlockUntilNextEventMatchingListInModeWithFilter
/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit +0x3997c _DPSNextEvent
/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit +0x38b18 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:]
/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit +0x2cf7c -[NSApplication run]
/opt/homebrew/Cellar/qt@5/5.15.10/plugins/platforms/libqcocoa.dylib +0x3d25c qt_plugin_instance
/opt/homebrew/Cellar/qt@5/5.15.10/lib/QtCore.framework/Versions/5/QtCore +0x1ca9a4 _ZN10QEventLoop4execE6QFlagsINS_17ProcessEventsFlagEE
/opt/homebrew/Cellar/qt@5/5.15.10/lib/QtCore.framework/Versions/5/QtCore +0x1cea24 _ZN16QCoreApplication4execEv
/Applications/klayout.app/Contents/Frameworks/libklayout_lay.0.dylib +0x7beb4 _ZN3lay14GuiApplication4execEv
/Applications/klayout.app/Contents/Frameworks/libklayout_lay.0.dylib +0x7a58c _ZN3lay15ApplicationBase3runEv
/Applications/klayout.app/Contents/MacOS/klayout +0x12ec8
/Applications/klayout.app/Contents/Frameworks/libklayout_rba.0.dylib +0x17c58 _ZN3rba15RubyInterpreter10initializeERiPPcPFiS1_S3_E
/opt/homebrew/Cellar/ruby/3.2.2_1/lib/libruby.3.2.dylib +0x1ce7e8 vm_call_cfunc_with_frame
/opt/homebrew/Cellar/ruby/3.2.2_1/lib/libruby.3.2.dylib +0x1d06f0 vm_sendish
/opt/homebrew/Cellar/ruby/3.2.2_1/lib/libruby.3.2.dylib +0x1b88e0 vm_exec_core
/opt/homebrew/Cellar/ruby/3.2.2_1/lib/libruby.3.2.dylib +0x1c7044 rb_vm_exec
/opt/homebrew/Cellar/ruby/3.2.2_1/lib/libruby.3.2.dylib +0x6f124 rb_ec_exec_node
/opt/homebrew/Cellar/ruby/3.2.2_1/lib/libruby.3.2.dylib +0x6ef98 ruby_run_node
/Applications/klayout.app/Contents/Frameworks/libklayout_rba.0.dylib +0x17be4 _ZN3rba15RubyInterpreter10initializeERiPPcPFiS1_S3_E
/Applications/klayout.app/Contents/MacOS/klayout +0x128d8
/Applications/klayout.app/Contents/MacOS/klayout +0x12444
/usr/lib/dyld +0x5f28 start
```
If I don't select the polygons and run that code , It could work well. But That code is a part of my macro, after flatten topcell, the macro will run `layout_view.get_selected` to get the shapes selected and do some calculate.
So is there a way to solve this problem?
Thanks,
Ethan Wang

klayoutmatthias, 2023-10-25:
I think, the problem happens because there are polygons selected which will be removed during the flatten. Maybe it is possible to avoid that problem, but it is a pretty basic one and there are too many ways to spoil things. 

To avoid this, you should call `LayoutView#cancel` before operations that potentially invalidate cell pointers or shapes. "cancel" will clear the selection and terminate potentially pending editor operations. 

Matthias

dsus4wang, 2023-11-02:
Ah after reading your answer, I clarified the logic of this operation. Thanks for your help!
wang

closed 2023-11-02 by dsus4wang

lukasc-ubc, 2023-11-17:
I've run into this issue a few times as well, enough that I've added it as a thing to look out for: https://github.com/SiEPIC/SiEPIC-Tools/wiki/Scripted-Layout#avoiding-klayout-crashes

Matthias, perhaps you could add a check that objects aren't selected before doing Python operations like delete, flatten, etc?  Or would that reduce performance?