KLayout/klayout #1516: About reading code related to DRC

gilgamsh, 2023-10-24:
I want to read the code about DRC in klayout, but I found it hard to read the code. 
I notice that in `src/db`, there is a file name box_tree. Does klayout use box_tree to do DRC? 
If so, what kind of space data structure is box_tree? 
I cannot find some material about it. Is that similar to quad tree?

klayoutmatthias, 2023-10-25:
`box_tree` is a quad tree, but that is not the core idea of the DRC feature.

Maybe you can enlighten me what is going on ... I receive a lot of similar questions recently. Apparently people study the code without much background knowledge and without much enthusiasm to understand complex concepts. Is that a kind of student project?

gilgamsh, 2023-10-26:
> `box_tree` is a quad tree, but that is not the core idea of the DRC feature.
> 
> Maybe you can enlighten me what is going on ... I receive a lot of similar questions recently. Apparently people study the code without much background knowledge and without much enthusiasm to understand complex concepts. I mean, code is always hard to read. Code is not a book.
> 
> Is that a kind of student project?

Thank you for you comments.  
So box tree is for gui?
It seems like in dbEdgesDelegate.h that you deal with DRC? Am i right?

Jebearssica, 2023-10-30:
> > > `box_tree` is a quad tree, but that is not the core idea of the DRC feature.
> > > Maybe you can enlighten me what is going on ... I receive a lot of similar questions recently. Apparently people study the code without much background knowledge and without much enthusiasm to understand complex concepts. I mean, code is always hard to read. Code is not a book.
> > > Is that a kind of student project?
> > 
> > 
> > Thank you for you comments. So box tree is for gui? It seems like in dbEdgesDelegate.h that you deal with DRC? Am i right?
> 
> are there any space data structures are used to do DRC?

Basically, the code from `db/` almost contains all the DRC involved data structures and classes, some of them are wrapper classes. Reading the code requires the background knowledge about IC design, such as you need to know what's the meaning of Layout/Cell/Instance... Also, Klayout is support for hierarchy, so you should not only familiar with the proper noun in flat design but also hierarchy design (such as hier-level/sub-instance/sub-cell...).

box tree is a helper class to store an object's bbox (Almost, and some of the object needs a BoxConvert to convert into box), which can efficiently search the related objects by box. Maybe kind of like KD-tree

gilgamsh, 2023-11-02:
> Basically, the code from `db/` almost contains all the DRC involved data structures and classes, some of them are wrapper classes. Reading the code requires the background knowledge about IC design, [...]

Thank you very much for your comments.

closed 2023-11-18 by gilgamsh