KLayout/klayout #1517: Calling `threads` in an XOR script, but it still doesn't use more than one thread

donn, 2023-10-25:
* Using KLayout 0.28.5

I typically use this script for XOR:

```sh
$ klayout -b -r ./xor.drc -rd top_cell=picorv32 -rd a=./picorv32.magic.gds -rd b=./picorv32.klayout.gds -rd jobs=16 -rd rdb_out=xor.xml -rd ignore=81/14
```

```rb
verbose

# Set up inputs
puts $a, $b
a = source($a, $top_cell)
b = source($b, $top_cell)

# Set up output
# target($gds_out, "XOR")
report("XOR #{$a} vs. #{$b}", $rdb_out)

def write_data(xor_data, layer_info)
  # xor_data.output(layer_info.layer, layer_info.datatype, layer_info.name)
  xor_data.output(layer_info.to_s, "XOR data for layer #{layer_info.to_s}")
end

# Run XOR
$jobs = $jobs.to_i or 1
info "Using #{$jobs} threads…"
threads($jobs)

## Collect all common layers
layers = {}
[ a.layout, b.layout ].each do |ly|
  ly.layer_indices.each do |li|
    i = ly.get_info(li)
    layers[i.to_s] = i
  end
end

ignore_list = $ignore.split(";")

## Perform per-layer XOR
total_xor_differences = 0
layers.keys.sort.each do |layer_name|
  if ignore_list.include? layer_name
    warn "Skipping #{layer_name}"
  else
    info "--- Running XOR for layer #{layer_name} ---"

    layer_info = layers[layer_name]
    xor_data = a.input(layer_name) ^ b.input(layer_name)
    total_xor_differences += xor_data.data.size
    info "XOR differences: #{xor_data.data.size}"

    write_data xor_data, layer_info
   end
end

info "---"
info "Total XOR differences: #{total_xor_differences}"

puts "%OL_CREATE_REPORT difference_count.rpt"
puts total_xor_differences
puts "%OL_END_REPORT"
```

Despite me setting `verbose`, using `psutils` or `top` shows that KLayout still uses a maximum of one thread. Am I missing something in this script?

---
I've attached a reproducible for your convenience: Just untar it and run `./run.sh`.
[one_thread.tar.gz](https://github.com/KLayout/klayout/files/13166148/one_thread.tar.gz)

klayoutmatthias, 2023-10-25:
That is because you're using flat mode. Flat mode is not threaded.

Tiled mode is (use `tiles(dim)` where "dim" is a good tile dimension - for a 130nm node for example, 500 to 1000 (µm) is a good tile size. But if your design is smaller than a single tile, multithreading won't happen too.

The XOR tool parallelizes layers too, but that option is not available in DRC as DRC statements executed linearly. Threading DRC statements would be an entirely different engine.

Matthias

donn, 2023-10-25:
ooooooh. gotcha. thanks for clarifying

closed 2023-10-25 by donn