KLayout/klayout #1519: Tips on speeding up build time?

donn, 2023-10-26:
On my Linux box, without `qtbindings`, KLayout takes 1h30m to build- This is the build invocation I'm using. Does anything immediately stand out as needing a fix..?

```sh
    CC=clang CXX=clang++\
      ./build.sh\
        -option -j$(nproc)\
        -without-qtbinding\
        -python $(which python3)\
        -ruby $(which ruby)\
        -prefix $out/lib\
        -expert\
        -verbose
```

klayoutmatthias, 2023-10-28:
What is nproc in your case?

1h30 feels very long. I tried a single-thread build myself with "-without-qtbinding" and it took 38m with gcc 11.4.0 on Ubuntu 22 on a i7-11800H @ 2.30GHz. Using all 8 cores takes the build time down to 8m.

I have seen significant differences in build times for different tool chains. KLayout makes use of complex templates in some places and some compilers have trouble compiling them quickly. I have also seen cases where linking was taking significant time and memory. The same step may take no time at all in other configurations.

I think that in compiler development, build performance usually is second priority over code performance as compiling is a one-time effort. So basically I'm not surprised to see excursions here. But multi-threaded building 

Matthias

stefanottili, 2023-10-28:
Just as another datapoint:
On a MacBook M1 (4+4 cores) using 8 threads it takes 7m30s to build klayout from scratch.

But I rarely do, a git pull followed by an (incremental) build only takes a couple tens of seconds.

donn, 2023-10-28:
Well, a 6-core machine (Ryzen 3600X).\