KLayout/klayout #1522: Changing ruler/annotation from script weirdly interferes with pya.Application.commit_config

EugeneZelenko, 2023-11-03:
We change default ruler/annotation from configuration script. But somehow calling `pya.Application.commit_config()` after that interfere with GUI appearance.

I tried KLayout versions 0.28.9 and 0.28.12 built in-house with Pythin 3.11.

KLayout command line is: `<path>\klayout -nc -rm ./configure.py -u <session path>`

Next configuration works fine (`Box` is title under icon and tool button menu items switched properly):

```
import pya

app = pya.Application.instance()

app.set_config('current-ruler-template', '7')
```

Next configuration is problematic (`Ruler` is text and selected item in menu, but actual annotation is box, choosing `Box` from menu leave both `Ruler` and `Box` checked in menu):

```
import pya

app = pya.Application.instance()

app.set_config('current-ruler-template', '7')
app.commit_config()
```

klayoutmatthias, 2023-11-07:
Hi Eugene,

I tried on Ubuntu, but I could not reproduce the behavior you report. Or maybe I just don't understand.

I will try on Windows myself. But if that problem is OS specific, maybe calling "app.process_events()" before "app.commit_config()" helps. 

Matthias

EugeneZelenko, 2023-11-07:
Hi, Matthias!

I tried to insert `app.process_events()` before `app.commit_config()` but it did not help. What is weirder, replacement of `app.commit_config()` with `app.process_events()` lead to same problem.

KLayout is built of RHEL 6 and run on Rocky Linux 8. Qt version is 5.14.2.

It may be useful to log configuration settings changes with some `-d` level. I tried `-d 200` but it looks like such changes were not reflected in log.

klayoutmatthias, 2023-11-08:
I think the log will not tell a lot.

Maybe we can clarify what I did to (not) reproduce the problem on Ubuntu 22.04:

* I opened some Layout in edit mode
* I selected "Ruler" mode with "Ruler" type selected

[image]

* I ran the script from above

```
import pya

app = pya.Application.instance()

app.set_config('current-ruler-template', '7')
app.commit_config()
```

Effect: the ruler switches to box mode. I can now start drawing a ruler and it becomes a box:

[image]

The menu is intact and I can switch to Ruler mode without issues.

What am I doing wrong?

Matthias

EugeneZelenko, 2023-11-08:
Hi, Matthias!

I'll try to dig deeper on my side, but it'll take some time.

EugeneZelenko, 2023-11-10:
Hi, Matthias!

I added debug prints after `PluginDeclaration::m_current_template` and `Service::m_current_template` updates.

Log for good configuration:

```
PluginDeclaration::PluginDeclaration 0
PluginDeclaration::configure 7
Service::Service 0
Service::configure 7
Service::Service 0
Service::configure 7
Service::Service 0
Service::configure 7
Service::Service 0
Service::configure 7
PluginDeclaration::config_finalize
PluginDeclaration::update_menu 7
PluginDeclaration::config_finalize
PluginDeclaration::config_finalize
PluginDeclaration::config_finalize
```

Log for bad configuration:

```
PluginDeclaration::PluginDeclaration 0
PluginDeclaration::configure 7
PluginDeclaration::config_finalize
PluginDeclaration::update_menu 7
Service::Service 0
Service::configure 7
Service::Service 0
Service::configure 7
Service::Service 0
Service::configure 7
Service::Service 0
Service::configure 7
PluginDeclaration::config_finalize
PluginDeclaration::update_menu 0
PluginDeclaration::config_finalize
PluginDeclaration::config_finalize
PluginDeclaration::config_finalize
```

klayoutmatthias, 2023-11-19:
Hi Eugene,

I did not pay attention to the "-nc" in your description. When I add that, I can reproduce the issue. I'll try to fix it.

Thanks,

Matthias

EugeneZelenko, 2023-11-20:
Hi, Matthias!

Thank you for help!

closed 2023-11-22 by klayoutmatthias