KLayout/klayout #1523: Missing Includes

wulfhednar02, 2023-11-06:
While building with Kubuntu 23.10, running the command
```
./build.sh -noruby -without-qt
```
causes a build error to occur. The error states missing definitions for <optional> and uint8_t. 

QMake suggests adding the following includes:
- GsiTypes.h
  - \<optional\>
  - \<cstdint\>
- GsiSerialization.h
  - \<optional\> 

After applying these changes to the header files, the compile successfully completes and the build begins. I suggest adding the includes to the header files as required.

klayoutmatthias, 2023-11-07:
Why should "<optional>" be required for "uint_8"? 

That is C++17 and so far, KLayout also compiles on C++11 with Qt < 6. That is important as there are many users out thers still using CentOS/RedHat 7 and even older.

Matthias

wulfhednar02, 2023-11-08:
\<optional\> is not for uint8_t. \<cstdint\> is a C++11 standard header for uint8_t. I am not sure about \<optional\> as you are correct in that is a C++17 requirement. I only know that QT6 would not compile it without the header.

So C++11 compiles that without the \<optional\> header just fine? Then just leave out that, but I believe the \<cstdint\> is still required.

wulfhednar02, 2023-11-12:
Closing issue as these have been looked into, and changes needed integrated into main.

closed 2023-11-12 by wulfhednar02