KLayout/klayout #1525: is klayout going to support Chinese Language?

wudenggang, 2023-11-06:
as the title

Kazzz-S, 2023-11-07:
Hi @wudenggang,

I think the answer would be **NO**.

----

Over nine years ago, I discussed the same topic (multi-language support) offline with Matthias.

I will not go into details, but I tried to translate only the menu/message strings (1,600+ items as of version 0.23.4) from English to Japanese and found the work challenging to accomplish.
[image]

If we expect ALL of them (including the assistant document), the amount of work will be enormous!

Kazzz-S

klayoutmatthias, 2023-11-07:
Ah yes. 

I'm not able to do that and after all, there is too much text and documentation to be updated and kept in sync once changes are made.

Matthias

s910324, 2023-11-08:
@Kazzz-S 

Is there a string/text resource file that can be translated to enable multi-language or you achieve this by simply grind throuth each and individual *.ui file?

Kazzz-S, 2023-11-08:
Hi @s910324,

It's the latter, basically.
It's such an old activity that I don't remember it very well.
However, I do remember suggesting the use of **QtLinguist**; the major problem was that due to the long development history of KLayout, not all related strings were included in the **QObject::tr()** function.

Kazzz-S