KLayout/klayout #1526: Cannot build klayout with python from anaconda3 on Mac M1

jasminabrar, 2023-11-06:
Hi, I am having difficulty building klayout from source code using anaconda3 python3.9. Specifically, I am getting the error message 'Unable to run python interpreter: /Applications/anaconda3/bin/python3.9' because my base conda environment is python3.11, thus python3.9 does not exist within the bin folder. If I activate a new python3.9 conda environment and attempt to run the build in there I continue to get this error because the macbuild script always looks for conda python3.9 at the /Applications/anaconda3/bin/python3.9 path (and when working in a new environment it is located at /Applications/anaconda3/envs/env_name/bin/python3.9Any help for how I could resolve this and successfully build klayout using conda would be appreciated. Thanks!

Kazzz-S, 2023-11-06:
Hi @jasminabrar,

As I do not have an M1 Mac, I cannot test this myself.  However, I assume that **Anaconda3** is also installed in **/Applications/anaconda3/** in an M1 environment.

Please try modifying the **build4mac_env.py** file as follows.
The latter is the one I've modified for the next maintenance release (0.28.13).

[image]
 |
to
 |
 v
[image]

Also, confirm that you have installed Ruby 3.1.4 with the following:
```
$ conda install ruby=3.1.4
```

Kazzz-S

jasminabrar, 2023-11-06:
@Kazzz-S Thank you for your response! I am encountering some errors when running the build and have been unsuccessful in resolving them. I was wondering if you have you encountered these errors before or if you have any suggestions to move forward. 

[image]

Something to note is that I am able to successfully build klayout using a fully homebrew flavoured build. I also have ruby 3.1.4 installed via conda.

Kazzz-S, 2023-11-07:
Hi @jasminabrar,

I believe your problem is related to https://bugreports.qt.io/browse/QTBUG-117225?filter=-4
The original warning was given by https://www.klayout.de/forum/discussion/2377/.
The solution is given in #1493. That is, please apply the patch to Qt5.

Kazzz-S

jasminabrar, 2023-11-07:
@Kazzz-S Thank you, that resolved my error. I am now getting a Project ERROR: Unknown module(s) in QT: designer. I encountered this error during an ubuntu build because one of the qt development tool packages was not installed, but am struggling to figure out what the cause of this error on the Mac might be. If you have any suggestions it would be greatly appreciated.

Kazzz-S, 2023-11-07:
Hi @jasminabrar,

I installed Anaconda3 from **Anaconda3-2023.09-0-MacOSX-x86_64.pkg** (for Intel Mac).
**QtDesigner** is there from the start, as shown below.
Maybe a packaging problem in the Apple Silicon version?

```
[Header Files]
(base) Ventura2{kazzz-s} qt (1)% pwd
/Applications/anaconda3/include/qt

(base) Ventura2{kazzz-s} qt (2)% ll | grep Designer
drwxr-xr-x  88 kazzz-s staff  2816 11  2 19:06 QtDesigner
drwxr-xr-x   7 kazzz-s staff   224 11  2 19:06 QtDesignerComponents

[Librafies]
(base) Ventura2{kazzz-s} lib (3)% pwd
/Applications/anaconda3/lib

(base) Ventura2{kazzz-s} lib (4)% ll | grep Designer
-rwxr-xr-x    2 kazzz-s staff   4304724  8 28 19:20 libQt5Designer.5.15.2.dylib
lrwxr-xr-x    1 kazzz-s staff        27 11  2 19:05 libQt5Designer.5.15.dylib -> libQt5Designer.5.15.2.dylib
lrwxr-xr-x    1 kazzz-s staff        27 11  2 19:05 libQt5Designer.5.dylib -> libQt5Designer.5.15.2.dylib
lrwxr-xr-x    1 kazzz-s staff        27 11  2 19:05 libQt5Designer.dylib -> libQt5Designer.5.15.2.dylib
-rw-r--r--    2 kazzz-s staff      1811  8 28 19:17 libQt5Designer.prl
-rwxr-xr-x    2 kazzz-s staff   1964064  8 28 19:20 libQt5DesignerComponents.5.15.2.dylib
lrwxr-xr-x    1 kazzz-s staff        37 11  2 19:05 libQt5DesignerComponents.5.15.dylib -> libQt5DesignerComponents.5.15.2.dylib
lrwxr-xr-x    1 kazzz-s staff        37 11  2 19:05 libQt5DesignerComponents.5.dylib -> libQt5DesignerComponents.5.15.2.dylib
lrwxr-xr-x    1 kazzz-s staff        37 11  2 19:05 libQt5DesignerComponents.dylib -> libQt5DesignerComponents.5.15.2.dylib
-rw-r--r--    2 kazzz-s staff      1905  8 28 19:17 libQt5DesignerComponents.prl
```

Kazzz-S

Kazzz-S, 2023-11-07:
Hi @jasminabrar,

[image]

What would happen if you turned off the **qtbinding**?
```
$ ./build4mac.py -q qt5ana3 -r ana3 -p ana3 --noqtbinding
```

Kazzz-S

jasminabrar, 2023-11-07:
@Kazzz-S Turning off qtbindings works to successfully build! Something to note, the -Y flag did not work for me as it caused an invalid code signature when I tried to open klayout, however the -y flag works fine. Thank you for all of your help!

To summarize for anyone who may come across this post later:

1. `$ ./build4mac.py -q qt5ana3 -r ana3 -p ana3 --noqtbinding`
2. `$ ./build4mac.py -q qt5ana3 -r ana3 -p ana3 --noqtbinding -y`

closed 2023-11-07 by jasminabrar